subsurface/subsurface #2200: iOS app crashes while synchonizing with the cloud

DivingSmurf, 2019-07-29:
### Describe the issue:

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
The iOS app crashes (=exits) now an then while sychronizing with the cloud. I'm using the manual cloud synchronization option.

### Operating system:

iOS environment: iPhone 6s Plus 64 GB with iOS 12.4
PC: Windows 10

### Subsurface version:

iOS: Subsurface Mobile 2.1.6
Windows PC: Subsurface 4.9.0

### Steps to reproduce:

 I'm not able to reproduce it reliable, but a good start might be:
1) Change a dive in the app and manuel sychronize with the cloud
2) Start Subsurface on a PC and change the same dive. Synchronize the cloud data after that
3) Sychronize the cloud data in the app 

### Current behavior:

App crashes.

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2019-07-29:
That is odd. If there is a synchronization issue at worst it should create odd artifacts in the data (which are usually easy to fix). I don't see why / how this could lead to a crash.
Does this happen every time you do that? I tried to reproduce this here and actually got the typical result for this scenario (one of the changes was dropped, the other prevailed).

DivingSmurf, 2019-07-29:
You misunderstood me. Or my description was bad. No change gets dropped (which is correct). What I do is:

1) I change an existing dive in the app. After that I start the manual cloud synchronization. I can exit the app now and go to another app, but I do not close the subsurface app completly.
2) I start subsurface on my pc. It automatically synchronizes the cloud data and loads the changes from step 1. Now I change the same dive from step 1 and start again the synchonization. Everything is correct until now and the dive was changed twice.
3) Now I start another manual cloud synchronization in the app. The expected result ist that the changes from step 2 are loaded. But the app crashes. After a restart the manual cloud synchronization works fine.

The app does not crash every time. I did not manage to reliably reproduce it.

Possibly something is not completly cleaned up in the background after sychronizing. After the after-crash-restarts everything works fine.

dirkhh, 2019-07-29:
Yes, that is indeed different from what I understood at first.
Especially since you say that things work as expected after you restart... that makes this something that is almost certainly about local state at runtime, not an issue with git data corruption (and I'm very happy about that - that could have been scary). Let me try this here and see if I can reproduce that issue
Thanks

mikeller, 2026-10-17:
Hi @DivingSmurf,

Thanks for reporting this. To look into the iOS crash while syncing with the cloud, we'd need a crash log (iOS usually captures one under Settings > Privacy & Security > Analytics & Improvements > Analytics Data) plus confirmation this still happens on a current build, cloud sync has changed substantially since 2019.

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.