subsurface/subsurface #2201: Crash on MacOS Mojave

fmatray, 2019-07-29:
### Describe the issue:
- [X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When I open Subsurface 4.9 on MacOS Mojave, the application crash with an error 'Bus error:10'

### Operating system:
MacOS Mojave 10.14.5 french version (up to date) on a macbook air 2015.

### Subsurface version:
Subsurface version : 4.9.0 downloaded from the website.
No problem with the previous version.

### Steps to reproduce:
1) /Applications/Subsurface.app/Contents/MacOS
2) ./Subsurface
Then crash.

### Current behavior:
a crash while opening cloud storage but not sure it's about this.

### Expected behavior:
well, no crash

### Additional information:
prometheus:MacOS $ ./Subsurface -v
Subsurface v4.9.0,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (ffbb47297539b873a1bc49d71ebd739639876b65)
built with Qt Version 5.11.1, runtime from Qt Version 5.11.1
built with libgit2 0.26.0
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): detected OpenGL version 2.1."
can't find Qt localization for locale "fr-FR" searching in "/Applications/Subsurface.app/Contents/translations"
cloud URL set as "https://cloud.subsurface-divelog.org//git/XXX@XXX.com[XXX@XXX.com]"
loading dive data from ("https://cloud.subsurface-divelog.org//git/XXX@XXX.com[XXX@XXX.com]")
git storage: Synchronisation du fichier de données
git storage: update local repo
sync with remote https://cloud.subsurface-divelog.org//git/XXX@XXX.com[XXX@XXX.com.com]
git storage: Synchroniser avec le stockage cloud
delete proxy setting
Cloud storage: checking connection to cloud server
Checking cloud connection...
git storage: fetch remote
git storage: Connecté au cloud, téléchargement des données distantes
git storage: check remote status
git storage: try to update
git storage: Synchronisation avec le stockage cloud effectuée
git storage: Chargement des plongées depuis le cache local
git storage: Succès de l'ouverture des données de plongée
**Bus error: 10**
prometheus:MacOS $ 

### Mentions:

dirkhh, 2019-07-29:
There is a potential crash in the map code that affects a very small subset of our users.
The latest development version has this problem fixed and we are planning to release 4.9.1 with that fix included within the next few days

dirkhh, 2019-07-29:
Please test this binary if you have a chance:
https://subsurface-divelog.org/downloads/test/Subsurface-4.9.0-35-g29f5d15a6692.dmg

fmatray, 2019-07-29:
it works. 
Thanks a lot for the quick answer.

closed 2019-07-29 by fmatray

dirkhh, 2019-07-29:
Excellent. Thanks for the report back!