subsurface/subsurface #2203: 4.9.0 for Windows crashes when synchronizing from cloud

estresao, 2019-07-30:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
After upgrading to Subsurface 4.9.0 on Windows 10 Enterprise it crashes loading data from cloud with no messages (just get closed)

### Operating system:

Windows 10 Enterprise, version 1709, OS Build 16299.1217 (English)

Laptop Lenovo T480
Processor Intel Core i7-8550U  @ 1.80Ghz 1.99Ghz
·2Gb RAM
System type: 64 bits

### Subsurface version:

4.9.0

No on my prevoius version (that is happening since I've upgraded to 4.9.0)

Official release 4.9.0

### Steps to reproduce:

1) Start up Subsurface
2) Open cloud storage
### Current behavior:

It seems to load and get closed
### Expected behavior:

Load data from cloud ant show it with no closing Subsurface

### Additional information:

### Mentions:

fmatray, 2019-07-30:
I have also a close issue, on Windows but with a fresh install (4.9.0) but no data on my computer (Windows 7 SP1 64 bits French).

The log with verbose :

C:\Users\GB5026\AppData\Local\Subsurface>subsurface.exe -v

C:\Users\GB5026\AppData\Local\Subsurface>
Subsurface v4.9.0,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (ffbb47297539b873a1bc49d71
ebd739639876b65)
built with Qt Version 5.12.0, runtime from Qt Version 5.12.0
built with libgit2 0.27.7

C:\Users\GB5026\AppData\Local\Subsurface>

estresao, 2019-07-30:
Adding log verbose following:

git storage: Sync with cloud storage
delete proxy setting
Cloud storage: checking connection to cloud server
Checking cloud connection...
git storage: fetch remote
git storage: Successful cloud connection, fetch remote
git storage: check remote status
git storage: try to update
git storage: Done syncing with cloud storage
git storage: Load dives from local cache
git storage: Successfully opened dive data

After that Subsurface windows get closed

chaosaffe, 2019-08-09:
Was able to repro on Ubuntu 18.04 with Subsurface v4.9.0. ending in a seg fault 

`segmentation fault (core dumped)  subsurface -v`

I'm not comfortable posting a core dump publically, but if one of the devs wants to take a look I'm happy to pass it over

<details>
 <summary>View Log</summary>

```bash
Subsurface v4.9.0,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (ffbb47297539b873a1bc49d71ebd739639876b65)
built with Qt Version 5.9.5, runtime from Qt Version 5.9.5
built with libgit2 0.26.0
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): detected OpenGL version 3.0."
loading dive data from ("/home/user/snap/subsurface/1596/.subsurface/user.xml")

File locations:

cloud URL set as "https://cloud.subsurface-divelog.org//git/not@real.io[not@real.io]"
Local git storage: /home/user/snap/subsurface/1596/.subsurface/cloudstorage/e54ac9c06463e71c
Cloud URL: https://cloud.subsurface-divelog.org//git/not@real.io[not@real.io]
Image filename table: /home/user/snap/subsurface/1596/.subsurface/hashes
Local picture directory: /home/user/snap/subsurface/1596/.subsurface/picturedata/

cloud URL set as "https://cloud.subsurface-divelog.org//git/not@real.io[not@real.io]"
Opening cloud storage from: "https://cloud.subsurface-divelog.org//git/not@real.io[not@real.io]"
git storage: Synchronising data file
git storage: update local repo
sync with remote https://cloud.subsurface-divelog.org//git/not@real.io[not@real.io]
git storage: Sync with cloud storage
delete proxy setting
Cloud storage: checking connection to cloud server
Checking cloud connection...
git storage: fetch remote
git storage: Successful cloud connection, fetch remote
git storage: check remote status
git storage: try to update
git storage: Done syncing with cloud storage
git storage: Load dives from local cache
git storage: Successfully opened dive data
[1]    5277 segmentation fault (core dumped)  subsurface -v
```

</details>

dirkhh, 2019-08-10:
Could you post the backtrace?

chaosaffe, 2019-08-11:
@dirkhh shared directly with you via Google Drive

mikeller, 2026-10-17:
Hi @estresao,

This report is from Subsurface 4.9.0 in 2019, and cloud sync has had a number of crash fixes since then that this report predates. With no further detail or confirmation this persists on a current build, we're closing it as stale.

Please reopen with a fresh crash log if you still see Windows crash while synchronizing from the cloud on a current release.

closed 2026-10-17 by mikeller