subsurface/subsurface #2204: Oceanic Geo 4.0 incorrect dive data

devinkr, 2019-07-30:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When downloading dives from the Oceanic Geo 4.0 the data is incorrect with the date for all dives showing as Jan 1, 1970 12:00 AM and depth, time, etc. are also incorrect.

### Operating system:
Windows 10 English

Yes, also happens on Android.

### Subsurface version:
4.9.0

Official release

### Steps to reproduce:

1) Select Import -> Import from dive computer
2) Select Geo 4.0
3) Click Download

### Current behavior:
Dives are downloaded showing the date of each dive as Jan 1, 1970 12:00 AM. After all dives have been download the downloaded dives list also show incorrect data.

### Expected behavior:
Expected behavior is that downloaded data matches that shown on dive computer logbook and accurately reflects actual dives.

### Additional information:

### Mentions:

sbogomolov, 2019-09-09:
I have exactly the same issue. I'm using the latest beta 2.2.0(4.9.1.40).

jefdriesen, 2019-10-19:
To investigate the problem, I need a memory dump of the dive computer. In the download dialog simply enable both the "Save libdivecomputer dumpfile" and  "Save libdivecomputer logfile" checkboxes. Attach both files to the bugreport.

devinkr, 2019-10-22:
When I check "Save libdivecomputer dumpfile" and "Save libdivecomputer logfile" and try to download the dive computer display freezes showing every icon and every digit as 88:88 etc. as in every segment of the 7-segment display is on. 

The download progress bar shows a percentage but doesn't show dives. It also doesn't show the model or serial number of the dive computer as it normally does. And then when it finally gets to 100% it says " Error No new dives downloaded from dive computer"

To unfreeze the dive computer I can turn off bluetooth on my laptop and it returns to normal.

I've attached the bin as a bin.txt as it wouldn't let me attach a .bin file.

[subsurfaceB.log](https://github.com/Subsurface-divelog/subsurface/files/3753464/subsurfaceB.log)

[subsurfaceB.bin.txt](https://github.com/Subsurface-divelog/subsurface/files/3753466/subsurfaceB.bin.txt)

If I don't check "Save libdivecomputer dumpfile" and "Save libdivecomputer logfile" and click download then it will show the model, and serial number and it will show the progress bar not with a percent but showing the current dive being downloaded but the date for each dive is shown as Jan 1, 1970 12:00 AM

jefdriesen, 2019-10-22:
The dive computer shouldn't freeze when downloading a memory dump. Does that also happen when downloading dives? In both cases the same communication is used, so there should be no difference in the behaviour. The fact that no dives are download is normal. The memory dump does only download the raw data, without any interpretation.

Can you download your dives with the Oceanic application and then export them to some human readable format (xml, csv, sqlite, etc). For decoding the data, having the already decode data (by the Oceanic application) to compare against helps a lot.

I have already found the most important fields, but not everything. You can check the progress by importing the attached xml file into subsurface. Carefully inspect the data, and report back any errors you find.

jefdriesen, 2019-10-22:
[geo40.zip](https://github.com/Subsurface-divelog/subsurface/files/3757441/geo40.zip)

devinkr, 2019-10-23:
The dive data in the xml looks good except for the pressures. There is no transmitter so there shouldn't be any cylinder pressure readings. The dates, times, depths, profiles, temperature, gas mix all look accurate.

The Oceanic application (mobile only) doesn't export anything which is why I want Subsurface to work. I can export from divecloud.net to Dan DL7 .zxu files but when I try to import those to Subsurface they always end up with things incorrect.

Here is that export for the same dives: [divecloud_20191023023438.zip](https://github.com/Subsurface-divelog/subsurface/files/3759724/divecloud_20191023023438.zip)

jefdriesen, 2019-10-23:
I have already fixed the tank pressure. The Geo 4.0 [specifications](https://www.oceanicworldwide.com/us/geo-4-0/) don't mention transmitter support at all, so I removed it.

You should be able to [sync](
https://www.oceanicworldwide.com/media/software/oceanic-computer-apps.png) the data from the mobile application to their desktop application (DiverLog). The DiverLog database is a sqlite3 database, which you can send directly without needing to export. It would be nice if you could get this info to fill in the missing pieces. But if not, no problem, because it looks like I already have the most of the info correct.

devinkr, 2019-10-23:
Unfortunately the only way to sync from mobile to the desktop application is if you purchase the Full version of the desktop application. 

I had a trial version at one point and synced some dives so I have attached what I think is the divelog file for some old dives. Note the serial number of the computer on these dives is different as I had to get a replacement Geo 4.0

[DiveLogSample.zip](https://github.com/Subsurface-divelog/subsurface/files/3760648/DiveLogSample.zip)

jefdriesen, 2019-10-23:
I didn't realize you need the full version for the syncing.

If the database file is from another dive computer, then I can't use it. It needs to match exactly with the dives in the memory dump. Otherwise I can't compare the data. But no worries I already pushed the fixes. They should appear in the next subsurface release.

mturkia, 2019-10-28:
> The Oceanic application (mobile only) doesn't export anything which is why I want Subsurface to work. I can export from divecloud.net to Dan DL7 .zxu files but when I try to import those to [...]

Can you describe, what is incorrect when you import the zxu files? I tested a few of them and to me it looked like they might be imported correctly. Well, there are a few negative temperatures, but they are in the zxu files. Anyway, if there is something wrong in the DL7 import, that should be fixed in Subsurface.

devinkr, 2019-10-28:
The depths were wrong and there was no temperature information. Instead of 100ft it was showing in subsurface as 100m. I just tried to re-import those dives and they appear to be correct now.

mikeller, 2026-10-17:
Hi @devinkr,

Thanks for all the data you provided for this. jefdriesen worked through your Geo 4.0's binary format field-by-field against the vendor's own DiverLog export, fixed the tank-pressure field confusion (the Geo 4.0 has no transmitter support at all per its spec), and pushed the fix, which you confirmed yourself: "I just tried to re-import those dives and they appear to be correct now."

Closing this as fixed and confirmed. Please reopen if you see incorrect dates, depths, or other fields importing from a Geo 4.0 on a current release.

closed 2026-10-17 by mikeller