subsurface/subsurface #2207: Trip date not correct on Trip notes page

djunkins, 2019-08-03:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When clicking on a trip in the dive list, the date in the Trip Notes page is not being updated to the selected trip. The trip date appears to remain on the date of the last individual dive selected even if that dive was not part of the trip.
### Operating system:

Mac OSX 10.14.5
### Subsurface version:

Tested on 4.9.0 and 4.9.1 and both exhibit this behavior.

### Steps to reproduce:

1) Select a dive in the dive list
2) Show the Notes page for the selected dive
3) Select a trip in the dive list
4) Observe that the date in the Trip Notes page remains displaying the date of the previously selected dive.

### Current behavior:

### Expected behavior:

The correct trip date should be displayed in the Trip Notes page.

### Additional information:

### Mentions:

bstoeger, 2019-08-04:
Thanks for the report - I will try to find the problem in the upcoming days.

djunkins, 2019-08-04:
I've been looking at it a bit, but I am having trouble. I would expect that in MainTab::updateDiveInfo() in maintab.cpp we would update the ui.dateEdit field somewhere around line 495. The problem is that there is no date feild in the "struct dive_trip" defined in core/trip.h. I guess we need to generate a date from the dives included in the trip?

bstoeger, 2019-08-07:
I have a patch that fixes this, will post a PR after dinner. For trips there is the trip_date() function.

PR #2215 (2019-08-08): Desktop: display trip time on main-tab
On the main tab, the trip time was not shown when switching to
a trip. Implement showing of the trip date in a function, as the
undo-code will also have to update the trip date in certain
circumstances.

Fixes #2207

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Fixes bug #2207 - not much more to say, really.
### Release note:

Done.
### Documentation change:
commit 344c85f4 (2019-08-08): Desktop: display trip time on main-tab

On the main tab, the trip time was not shown when switching to
a trip. Implement showing of the trip date in a function, as the
undo-code will also have to update the trip date in certain
circumstances.

Fixes #2207
closed 2019-08-08 by dirkhh