subsurface/subsurface #2210: Crash on launch (Windows 7 + SubSurface 4.9.1)

fmatray, 2019-08-05:
### Describe the issue:
- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
SubSurface 4.9.1 crash on launch.

### Operating system:
Windows 7 Enterprise SP1 64bits, french version
CPU: Intel i5-6200 2.3GHz
RAM: 16Go

### Subsurface version:
4.9.1 with a fresh install

### Steps to reproduce:
C:\Users\XXXXX\AppData\Local\Subsurface>subsurface.exe -v -v -v

C:\Users\XXXXX\AppData\Local\Subsurface>
Subsurface v4.9.1,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (ffbb47297539b873a1bc49d71
ebd739639876b65)
built with Qt Version 5.12.0, runtime from Qt Version 5.12.0
built with libgit2 0.27.7

Then I have a windows crash message.

### Current behavior:
Crash without a clear message

### Expected behavior:
Well, no crash :-)

### Additional information:
Message from Windows (sorry, it's only in french)

Signature du problème :
  Nom d’événement de problème:	APPCRASH
  Nom de l’application:	subsurface.exe
  Version de l’application:	0.0.0.0
  Horodatage de l’application:	00000000
  Nom du module par défaut:	ig9icd32.dll
  Version du module par défaut:	21.20.16.4821
  Horodateur du module par défaut:	59d2911b
  Code de l’exception:	c0000005
  Décalage de l’exception:	004eaf36
  Version du système:	6.1.7601.2.1.0.256.4
  Identificateur de paramètres régionaux:	1036
  Information supplémentaire n° 1:	0a9e
  Information supplémentaire n° 2:	0a9e372d3b4ad19135b953a78882e789
  Information supplémentaire n° 3:	0a9e
  Information supplémentaire n° 4:	0a9e372d3b4ad19135b953a78882e789

### Mentions:

fmatray, 2019-08-05:
I have the same problem with subsurface 4.8.6. So, it can be a problem with my computer.

Do you have a compiled version for windows with debug symbols ?

bstoeger, 2019-08-06:
Could you perhaps send your log to @dirkhh?
Is there any version that works?

neolit123, 2019-08-11:
googling ig9icd32.dll reveals relation to the Intel Graphics driver chipset.
have you installed the latest drivers for your motherboard/GPU?

```
C:\Users\XXXXX\AppData\Local\Subsurface>subsurface.exe -v -v -v

C:\Users\XXXXX\AppData\Local\Subsurface>
Subsurface v4.9.1,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (ffbb47297539b873a1bc49d71
ebd739639876b65)
built with Qt Version 5.12.0, runtime from Qt Version 5.12.0
built with libgit2 0.27.7
```

normally running: `subsurface.exe -v -v -v` should dump GPU related information
are you sure this is the whole log?

alternatively, what happens if you try the following:
```
set QT_QUICK_BACKEND=software
```

before running subsurface.exe from the console?

fmatray, 2019-08-12:
Using  'set QT_QUICK_BACKEND=software' works nicely. Thanks a lot.

I can't update the drivers as I'm not the admin of this computer (I'm at work).
I sent  all logs I have. There is no other GPU dump.

Thanks.

closed 2019-08-12 by fmatray

neolit123, 2019-08-12:
please note that `set QT_QUICK_BACKEND=software` renders the Map widget without GPU acceleration, so it will be slow.