subsurface/subsurface #2212: Planner / profile hang

bstoeger, 2019-08-05:
[image]

### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I can easily make the parser hang on save:
Create a crazy profile such as in the attached screen shot. Note the crazy cylinder values. In save the application hangs. This happens on an empty dive log. I'm not sure if this is relevant.

### Operating system:

Kubuntu 19.4

### Subsurface version:

Current master.
### Steps to reproduce:

1) Open empty log
2) Create crazy file as per attached screenshot
3) Save
### Current behavior:

Hang
### Expected behavior:

No hang
### Additional information:

Backtrace:
#0  0x00007ffff2f39f8d in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007ffff2f3cabe in QObject::connect(QObject const*, char const*, QObject const*, char const*, Qt::ConnectionType) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#2  0x00005555557c5e00 in DiveTextItem::setText (this=this@entry=0x55557bbffba0, t=...)
    at /home/bs/src/subsurface/profile-widget/divetextitem.cpp:63
#3  0x00005555557b9019 in DiveCartesianAxis::updateTicks (this=0x5555567f2030, color=TIME_GRID)
    at /home/bs/src/subsurface/profile-widget/divecartesianaxis.cpp:207
#4  0x00005555557b7857 in DiveCartesianAxis::animateChangeLine (this=0x5555567f2030, newLine=...)
    at /home/bs/src/subsurface/profile-widget/divecartesianaxis.cpp:266
#5  0x00005555557a464a in ProfileWidget2::setProfileState (this=0x55555669c140) at /home/bs/src/subsurface/profile-widget/profilewidget2.cpp:1236
#6  0x0000555555691392 in MainWindow::showProfile (this=0x555555da6630) at /home/bs/src/subsurface/desktop-widgets/mainwindow.cpp:744
#7  0x00005555556913e9 in MainWindow::planCreated (this=0x555555da6630) at /home/bs/src/subsurface/desktop-widgets/mainwindow.cpp:830
#8  0x00005555556dfa88 in MainWindow::qt_static_metacall (_o=0x555555da6630, _c=<optimized out>, _id=<optimized out>, _a=<optimized out>)
    at /home/bs/src/subsurface/build/desktop-widgets/subsurface_interface_autogen/EWIEGA46WW/moc_mainwindow.cpp:380
#9  0x00007ffff2f37426 in QMetaObject::activate(QObject*, int, int, void**) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00005555557e614b in DivePlannerPointsModel::createPlan (this=0x555555b2cf00 <DivePlannerPointsModel::instance()::self>, 
    replanCopy=<optimized out>) at /home/bs/src/subsurface/qt-models/diveplannermodel.cpp:1143
#11 0x00005555557ff577 in DivePlannerPointsModel::qt_static_metacall (_o=0x555555b2cf00 <DivePlannerPointsModel::instance()::self>, 
    _c=<optimized out>, _id=<optimized out>, _a=0x7fffffffcc70)
    at /home/bs/src/subsurface/build/qt-models/subsurface_models_desktop_autogen/EWIEGA46WW/moc_diveplannermodel.cpp:321
#12 0x00007ffff2f37426 in QMetaObject::activate(QObject*, int, int, void**) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff46717e0 in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff2f37426 in QMetaObject::activate(QObject*, int, int, void**) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff45d2032 in QAbstractButton::clicked(bool) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff45d224a in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff45d360f in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff45d37e5 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff4528b58 in QWidget::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff44e9551 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff44f0b77 in QApplication::notify(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff2f0d8e9 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007ffff44efebf in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff454357b in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff454653f in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007ffff44e9551 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#27 0x00007ffff44f0930 in QApplication::notify(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00007ffff2f0d8e9 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#29 0x00007ffff3a48c6c in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#30 0x00007ffff3a4a075 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#31 0x00007ffff3a2405b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#32 0x00007fffec32167a in ?? () from /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#33 0x00007ffff0f889ee in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#34 0x00007ffff0f88c88 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#35 0x00007ffff0f88d1c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#36 0x00007ffff2f61047 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#37 0x00007ffff2f0c5bb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#38 0x00007ffff2f145e2 in QCoreApplication::exec() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
--Type <RET> for more, q to quit, c to continue without paging--
#39 0x00005555556847e0 in main (argc=<optimized out>, argv=<optimized out>) at /home/bs/src/subsurface/subsurface-desktop-main.cpp:116

### Mentions:

@atdotde

atdotde, 2019-08-06:
I cannot reproduce this. Unfortunately, you screenshot does not show your "crazy cylinder values".

atdotde, 2019-08-06:
OK, now, I get it: Your cylinder is so small that the end pressure under-runs the minimal value of cylinder.end.mbar which is a pressure_t which is an int32, so it goes to about -2000bar before under-running.

I see two possible approaches to a fix:

a) Don't let the planner go to negative pressures. When the cylinder is empty, it's empty. This amounts to adding

`if (cyl->end.mbar < 0)
		cyl->end.mbar =0;`

in the end of update_cylinder_pressure in planner.c.

Disadvantage: You only get your gas use in volume units not a reasonable negative pressure (which would inform you how many more cylinders you would need. A somewhat more pragmatic approach would be to allow some negative values but not crazy negative, for example limit to minus working pressure.

b) give more bits to the end pressure. I tried making pressure_t an int64 but that was not enough. I guess one would also need to look at the code that does the pressure interpolation to use long all along.

Any opinions?

torvalds, 2019-08-07:
I wrote a longer email on the list. No, pressure_t isn't the problem. 

I suspect some intermediate calculation multiplies depth (in mm) and pressure (in mbar) together (or possibly cylinder volume in mm), and doesn't reduce ranges, so when while both values _comfortably_ fit in their respective types, the intermediate value from the multiplication of two 32-bit values will overflow. And while it gets converted back to some pressure later (possibly by just fixing up the range by a fixed unit divisior), by that time the overflow has long since happened already and the end result is random.

torvalds, 2019-08-07:
Some of the craziness is mitigated by

   https://github.com/Subsurface-divelog/subsurface/pull/2214

but that's really only band-aid and makes sure that when the planner starts using crazy values, the gas compressibility code doesn't then push the crazy values even further.

The real fix is to make the planner not use negative gas pressures. So I'm not closing this bug report.

atdotde, 2019-08-08:
I believe #2214 is actually a fix for the problem. As said before, negative pressures make sense as they inform the user how much gas is actually missing.

PR #2217 (2019-08-08): Negative Pressure

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [x] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

In the planner, it is possible to plan dives that use more gas than available. In this case, we end up with negative pressures in the cylinder. This information is useful as it gives a rough estimate of how many more cylinders are required.

To alert the user of this fact, this patch colours the pressure line in a highly visible color when this happens.

 [... cut]
closed 2019-08-08 by dirkhh