subsurface/subsurface #2221: Android crash in DiveImportedModel::repopulate

dirkhh, 2019-08-09:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Backtrace:
```
backtrace:
  #00  pc 00000000001798aa  /data/app/org.subsurfacedivelog.mobile-Ac2S9LYxrLIi7igc4P96lQ==/lib/arm/libsubsurface-mobile.so (DiveImportedModel::repopulate(dive_table*, dive_site_table*)+33)
  #01  pc 0000000000183989  /data/app/org.subsurfacedivelog.mobile-Ac2S9LYxrLIi7igc4P96lQ==/lib/arm/libsubsurface-mobile.so (DiveImportedModel::qt_metacall(QMetaObject::Call, int, void**)+52)
  #02  pc 00000000000f673b  /data/app/org.subsurfacedivelog.mobile-Ac2S9LYxrLIi7igc4P96lQ==/lib/arm/libQt5Qml.so
```

### Operating system:

Most of the crash reports (more than a hundred) are on Android 8.1 and 9

### Subsurface version:

4.9.0.35

### Mentions:

#2219 was supposed to fix this one - and I haven't seen this crash reported in the latest build based on #2219

bstoeger, 2019-08-10:
Oh my ... These backtraces aren't very helpful. :(
We seem to have a very fundamental problem, but how to debug this? I reckon this can't be reproduced on mobile-on-desktop...?

bstoeger, 2019-08-10:
The fix makes no sense to me. The call in `mobile-widgets/qml/DownloadFromDiveComputer.qml` is
```
importModel.repopulate(table, sites)
```
`table` is a property of `DownloadThread` connected to the `table()` function:
```
struct dive_table *DownloadThread::table()
{
        return &downloadTable;
}
```
Note that the return of this function can *never* be null.
So how could a null check fix this?
The only way we could get a null here is that somehow the connection between QML and C++ objects isn't established yet. But we are inside the QML object, so apparently it exists. Something is completely broken here.

As for the fix - it seems more sensible to do the check in QML [if (table && site) ...] and perhaps give an error message.

Is there any description of what the users did to reach this error?

dirkhh, 2019-08-10:
I pasted all the information that I have.
These are automatic crash reports. No user notes, not even an ability to identify the user and ask them.
The comments in the negative reviews fall into two categories:
- crash right after the app starts
- crash after downloading from dive computer

bstoeger, 2019-09-14:
This is horribly reminiscent of the recent crashes we had: A model is reset in QML context.

I'd really like to remove the code from DownloadFromDiveComputer.qml

@dirkhh: is there an easy way to test download on mobile-on-desktop. I faintly remember that Jef mentioned a dive-computer simulator?

dirkhh, 2019-09-14:
Yes there is. It's a bit finicky to set up, but once you fought through the first pain, it's really not that complicated

bstoeger, 2019-09-15:
I downloaded it and it seems to run fine. Would you have an interesting dump which I could use?

dirkhh, 2019-09-15:
You can create your own by creating a libdivecomputer dump file when downloading from within Subsurface...
I have a few here if you want to try mine, but it's really straight forward and then you know what should be in the dump :-)

closed 2019-10-28 by dirkhh