subsurface/subsurface #2224: Android crash in trip_shown_dives

dirkhh, 2019-08-09:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

backtrace
```
  #00  pc 00000000002a8abc  /data/app/org.subsurfacedivelog.mobile-_noeo-4a2W3q-VScJ2MvdQ==/lib/arm64/libsubsurface-mobile.so (trip_shown_dives+64)
  #01  pc 000000000023211c  /data/app/org.subsurfacedivelog.mobile-_noeo-4a2W3q-VScJ2MvdQ==/lib/arm64/libsubsurface-mobile.so (DiveListSortModel::tripTitle(QVariant const&)+192)
  #02  pc 0000000000237d9c  /data/app/org.subsurfacedivelog.mobile-_noeo-4a2W3q-VScJ2MvdQ==/lib/arm64/libsubsurface-mobile.so
  #03  pc 0000000000238024  /data/app/org.subsurfacedivelog.mobile-_noeo-4a2W3q-VScJ2MvdQ==/lib/arm64/libsubsurface-mobile.so (DiveListSortModel::qt_metacall(QMetaObject::Call, int, void**)+76)
  #04  pc 00000000002ffe3c  /data/app/org.subsurfacedivelog.mobile-_noeo-4a2W3q-VScJ2MvdQ==/lib/arm64/libQt5Qml.so
```

### Operating system:

Android 8.1

### Subsurface version:

4.9.1.16 based on #2219

bstoeger, 2019-08-10:
This one here makes me very nervous.

In tripTitle() we explicitly check that trip is a valid reference. And still it crashes. This means that the core data structures are corrupted.

dirkhh, 2019-08-10:
I can't run Valgrind on Android. I can run Valgrind on mobile-on-Desktop and am happy to try this again, frustrating as it generally appears to be.
Family appears to be taking up most of my time today - I'll try to get to this, soon.

bstoeger, 2019-08-10:
Sadly, the last time I tried to find a data corruption bug (loading a log with many weightsystems on old version), valgrind didn't help.

bstoeger, 2019-08-11:
I tried to run mobile-on-desktop under valgrind and didn't manage to hit any invalid memory accesses (apart from the usual rendering stuff). I added, removed, edited dives and used the undo functionality. Moreover I didn't see anything fundamentally wrong in the dive-manipulation code of qmlmanager.cpp.

I must say that I'm quite weary of DiveObjectHelper - this represents a reference to a dive and we have no control over it. With that I mean: if we delete a dive, how can we know that there are no more of the DiveObjectHelpers flying around? But I can't point to a place in the code where I could say that this is wrong. In the long run I'd like to change the way QML and core interact to use models and no references (ideally those models that are also used by desktop). But in the short run I'm a bit stumped.

dirkhh, 2019-08-11:
We changed the way core and QML interact twice already. And I'm with you, I hate this code.
But I have spent quite some time trying the same (valgrind) with no luck.
On the flip side, no crashes reported, yet, with my "bogus" fixes.

bstoeger, 2019-09-14:
Ok - this appears to be the horrible pass pointer to trip via its hex-representation. This will have to go.

commit 57149797 (2019-08-09): Avoid potential crash when accessing the dives in a trip

This should never happen as all dives should be initialized, but we see this
as a crash location on Android.

Fixes #2224
closed 2019-10-28 by dirkhh