subsurface/subsurface #2225: Android crash in DiveObjectHelper::tripNrDives

dirkhh, 2019-08-09:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

backtrace
```
#00  pc 00000000002e76c0  /data/app/org.subsurfacedivelog.mobile-i3h-oZyARGZtxq5wXe9HWw==/lib/arm64/libsubsurface-mobile.so (DiveObjectHelper::tripNrDives() const+12)
  #01  pc 00000000002fd338  /data/app/org.subsurfacedivelog.mobile-i3h-oZyARGZtxq5wXe9HWw==/lib/arm64/libsubsurface-mobile.so
  #02  pc 00000000001d4398  /data/app/org.subsurfacedivelog.mobile-i3h-oZyARGZtxq5wXe9HWw==/lib/arm64/libQt5Qml.so (QV4::QObjectWrapper::getProperty(QV4::ExecutionEngine*, QObject*, QQmlPropertyData*)+728)
```
### Operating system:

Android 9

### Subsurface version:

4.9.1.16 based on #2219

dirkhh, 2019-08-09:
This one is nuts. This should only be able to crash if a `DiveObjectHelper` was created for an invalid dive - and in that case it should crash right in the constructor of `DiveObjectHelper`.
Not sure what to do about this one.

bstoeger, 2019-08-10:
Indeed, this very much looks like data corruption. Could be use-after-free or write-beyond-end-of-array.

bstoeger, 2019-08-14:
There is one function that is *very* scary:

```
// In QML, section headings can only be strings. To identify dives that
// belong to the same trip, a string containing the trip-pointer in hexadecimal
// encoding is passed in. To format the trip heading, the string is then
// converted back with this function.
QVariant DiveListSortModel::tripIdToObject(const QString &s)
{
        if (s.isEmpty())
                return QVariant();
        return QVariant::fromValue((dive_trip *)s.toULongLong(nullptr, 16));
}
```

This transforms a string into a pointer. Who knows if this works on ARM64 just as on x86-64? This one will have to go (and yes, I fear that I came up with that craziness).

dirkhh, 2019-08-15:
That's kinda cute, isn't it? And not fragile at all.
Looking at it I wonder why we decided to do it that way back then.

bstoeger, 2019-08-15:
It's my fault - the original code used to format all string data into a long string and decomposed that in the "tripHeaderDelegate". I thought it was too wasteful to construct that monster string for every dive, when it should only be needed once per trip.

There seems to be no sensible way to pass data down to the section-header. If you try to pass anything other than a string (e.g. a trip-object) the thing fails, because it attempts to do a case-insensitve string comparison. That is obviously only supported for string types. What a disaster.

Probably we should switch from ListView to TreeView, but I am *not* looking forward to do that.

dirkhh, 2019-08-15:
Snort.
I remember how much fun it was to get trips to work in QML in the first place.
May I humbly suggest that my solution with having this string created for every dive maybe wasn't great, but that it had the advantage of working?

bstoeger, 2019-08-15:
I'm quite sure that the current version works as well - at least I haven't seen any crash report that would suggest that it doesn't (note that the generated trip pointer is only read from, so it can't produce data corruption). This was more an idle remark and I shouldn't have posted it here.

It's just too ugly to continue existing. Is there any reason that we can't use TreeView instead of ListView (apart from the pain debugging anything QML)?

bstoeger, 2019-09-14:
Closing, since the `DiveObjectHelper::tripNrDives()` function in the backtrace doesn't exist anymore. Of course this doesn't necessarily mean that the root cause of the data corruptions has been fixed.

closed 2019-09-14 by bstoeger