subsurface/subsurface #2226: Error with Cache Data moving for Cloud Storage

djunkins, 2019-08-09:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When opening Subsurface on Mac OS X with the current master (through commit 302b39b), a red error bar appears at the bottom saying "Moved cache data..." with the new cache file have a integer appended to it (e.g. 1f192ae9e4ba4a5f.2). When the user clears the error, no log is loaded. If the "Open Cloud Storage" menu option is clicked, the log loads normally, but when a change made to the log and the "Save to cloud storage" option is clicked, another error bar appears saying "Unable to open git repository ...". After clearing the second error message, an second attempt to save to the cloud storage appears to work. 

This issue has appeared since the last time I build the master branch which was on commit 78f425d.

### Operating system:

Mac OSX 10.14.6

### Subsurface version:

Subsurface master branch (git hash 302b39bcd56c85bfacd0b4cbc48e9044a3908bca)
[image]
[image]

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

bstoeger, 2019-08-10:
> This issue has appeared since the last time I build the master branch which was on commit 78f425d.

Does that mean that 78f425d still works? Because I would have guessed that the error was introduced in the "make git parser reentrant" searies. But 78f425d is *after* that. It's only a handful of commits between the two that you list. Could you perhaps git-bisect it? Thanks.

djunkins, 2019-08-10:
@bstoeger Correct. I did not receive the error on commit 78f425d. I started to try to figure out which commit caused it, but had other commitments yesterday. I'll continue to try to isolate it today.

djunkins, 2019-08-11:
Here is the results of using git bisect between the known good and bad commits:

```
orcas:subsurface junkins$ git bisect good
594d1d3514f91221884bdc31f29a09e8f325f1e8 is the first bad commit
commit 594d1d3514f91221884bdc31f29a09e8f325f1e8
Author: Berthold Stoeger <bstoeger@mail.tuwien.ac.at>
Date:   Mon Aug 5 20:07:10 2019 +0200

    Cleanup: move file-related function declarations to file.h
    
    A number of architecture-dependent functions were declared in
    dive.h. Move them to file.h so that not all file-manipulating
    translation units have to include dive.h. This is a small step
    in avoiding mass-recompilation on every change to dive.h
    
    Signed-off-by: Berthold Stoeger <bstoeger@mail.tuwien.ac.at>

:040000 040000 0e92d4f69494994c572cea923afdcabfa3c50a5035eed219822e2fc5e1ff515a1310ae7bd6cc93e3 M	core
:040000 040000 48dff0e6338de68f2cb7178184a96194cd4f41521ebc2a8e8d0168193425093f485ee5899e28eea1 M	desktop-widgets
orcas:subsurface junkins$ 
```

I noticed that there were some warnings in the compile of macos.c that were not present in the earlier commits that worked:

```
/Users/junkins/Source/subsurface/core/macos.c:199:46: warning: declaration of 'struct stat' will not be visible outside of this function [-Wvisibility]
int subsurface_stat(const char* path, struct stat* buf)
                                             ^
/Users/junkins/Source/subsurface/core/macos.c:201:9: warning: implicit declaration of function 'stat' is invalid in C99 [-Wimplicit-function-declaration]
        return stat(path, buf);
               ^
/Users/junkins/Source/subsurface/core/macos.c:206:9: warning: implicit declaration of function 'zip_open' is invalid in C99 [-Wimplicit-function-declaration]
        return zip_open(path, flags, errorp);
               ^
/Users/junkins/Source/subsurface/core/macos.c:206:9: warning: incompatible integer to pointer conversion returning 'int' from a function with result type 'struct zip *' [-Wint-conversion]
        return zip_open(path, flags, errorp);
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/junkins/Source/subsurface/core/macos.c:211:9: warning: implicit declaration of function 'zip_close' is invalid in C99 [-Wimplicit-function-declaration]
        return zip_close(zip);
               ^
5 warnings generated.

```

PR #2233 (2019-08-11): Add missing header files to macos.c
Missing header files were causing errors opening cloud storage on
launch on Mac OSX.

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

macos.c was missing zip.h and sys/stat.h when the file functions were moved from dive.c to architecture specific files.

### Changes made:

### Related issues:

Fixes #2226 

### Additional information:

### Release note:
closed 2019-08-11 by bstoeger