subsurface/subsurface #2235: Android crash in get_dive_location

dirkhh, 2019-08-11:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Another crazy crash that should be impossible
backtrace
```
#00  pc 0000000000243d10  /data/app/org.subsurfacedivelog.mobile-Sc1HZzdPAzRd2Z-FKQs06w==/lib/arm64/libsubsurface-mobile.so (get_dive_location+8)
  #01  pc 00000000002e4fb0  /data/app/org.subsurfacedivelog.mobile-Sc1HZzdPAzRd2Z-FKQs06w==/lib/arm64/libsubsurface-mobile.so (DiveObjectHelper::location() const+24)
  #02  pc 00000000002fd030  /data/app/org.subsurfacedivelog.mobile-Sc1HZzdPAzRd2Z-FKQs06w==/lib/arm64/libsubsurface-mobile.so
  #03  pc 00000000001d4600  /data/app/org.subsurfacedivelog.mobile-Sc1HZzdPAzRd2Z-FKQs06w==/lib/arm64/libQt5Qml.so (QV4::QObjectWrapper::getProperty(QV4::ExecutionEngine*, QObject*, QQmlPropertyData*)+1344)
```
I am beginning to wonder if QML somehow accesses a `DiveObjectHelper` object that wasn't created from a valid dive...

### Operating system:

Android 9 on a Samsung Galaxy S8

### Subsurface version:

4.9.1.18 corresponding to #2219

bstoeger, 2019-08-12:
Can you disassemble get_dive_location, so that we can see if `dive` or `dive->dive_site` is invalid? get_dive_location+8 is *very* early in the function so I would guess that dive is invalid, but as you say, that makes no sense, since the constructor of DiveObjectHelper should have crashed.

bstoeger, 2019-08-12:
Ok, after looking at divelistmodel.cpp I have a somewhat plausible theory of what's going on:

Note how the DiveListModel stores pointers-to-DiveObjectHelper. Moreover we do not pass DiveObjectHelpers, but pointers-to-DiveObjectHelper through QML.

In `DiveListModel::removeDive(int i)` the DiveObjectHelper is deleted. Who guarantees that all the pointers in QML are also gone? This all looks to me like use-after-free of DiveObjectHelper, not of the dive itself.

I think we were looking at the wrong way: not the dive is invalid, but the DiveObjectHelper.

dirkhh, 2019-08-12:
I bet it's those pesky C programmers passing pointers instead of objects. This sounds credible to me. It would explain what we see - there are two other crashes of exactly the same type that I didn't even create issues for because they also clearly come from invalid dives in a DiveObjectHelper - or as you point our, from invalid DiveObjectHelpers

bstoeger, 2019-08-13:
For testing, I turned DiveObjectHelper into a Q_GADGET based class and pass it around as value. The bad news is that now I get crashes. The good news is that it is *reproducible* crashed, so we might have a chance at debugging them.

dirkhh, 2019-08-13:
I like how you see the positive in these things.
I will be traveling like crazy for 16 days starting today. I will try to stay on top of things as much as I can, but there's a chance that I'll be sometimes a bit slower to review / test things.

bstoeger, 2019-09-14:
Closing, since the `DiveObjectHelper::location()` function in the backtrace doesn't exist anymore. Of course this doesn't necessarily mean that the root cause has been fixed.

closed 2019-09-14 by bstoeger