subsurface/subsurface #2236: Cannot open existing local git repo

willemferguson, 2019-08-12:
When saving a local git repo, one can afterwards go to File->Open and the git repo is in the list of recently used dive logs. However, if one exits Subsurface and restarts it, the repo is not listed in the list of recently used dive logs any more. When trying to open an existing repo, there is (at least in Gnome) not a facility to type any file name. This makes it impossible to open the repo after saving it.

### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

Subsurface V 4.9.1 on Ubuntu 18.04

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@dirkhh

dirkhh, 2019-08-12:
I can confirm both parts of this.
Admittedly, local git repos aren't really a "user" feature, but we should try to make this work better.
The Gnome File-Open dialog that doesn't offer an entry field for the file name is... __special__
The fact that we don't keep the repo around as "recent" file is the most annoying in all the annoying things here.
The immediate workaround is to start Subsurface from the command line like this
`subsurface <path>/<folder>/[branch]`
i.e. in my example
`subsurface ~/.subsurface/localrepo/\[test\]`
(you need the backslashes to avoid the shell thinking that these are file glob markers)

dirkhh, 2019-08-12:
I forgot - the "not having a file name entry field" is considered a feature... this makes opening repos from within the app impossible. But at least now they are stored in recent files

PR #2237 (2019-08-18): Desktop: show local git repos in recent files
But don't show our cloud storage entry (as that is already in the File menu,
anyway).

This is extremely useful because while you can manually enter a file name to
save to (and therefore can use the 'magic' git repo syntax), on most OSs there
is no way to enter that non-existing 'file name' (which is the git branch in
square brackets) in the file open dialog.

Fixes: #2236

### Describe the pull request:

- [ ] Bug fix
- [x] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language tr [... cut]
commit 5ffca686 (2019-08-18): Desktop: show local git repos in recent files

But don't show our cloud storage entry (as that is already in the File menu,
anyway).

This is extremely useful because while you can manually enter a file name to
save to (and therefore can use the 'magic' git repo syntax), on most OSs there
is no way to enter that non-existing 'file name' (which is the git branch in
square brackets) in the file open dialog.

Fixes: #2236
closed 2019-08-18 by neolit123