subsurface/subsurface #2245: Failure in the import from the i77or

wegi82, 2019-08-13:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Today after the dive i synced my Aqualung i770r with the subsurface app on android. The dive was 14.3m deep. Subsurface shows 24m. 

The dive two day ago was 23.5m and subsurface shows 24m.

I tried also the diverlog+ app. There is the deep correct with 14.33m.

Now i have tried it with the cable and als subsurface on windows tells me 24m.

### Operating system:

windows 7, Android version 9

Aqualung i770r Firmware 1D.02

### Subsurface version:
 4.9.1

No

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mturkia, 2020-02-04:
Sorry for long delay in response. Internally Subsurface stores and uses exact depths as was imported, but rounded numbers are displayed for user. If the depth is less than 20m, one decimal is shown (if I recall correctly), for deeper dives only full meters are displayed.

However, the following statement cannot be explained by this. Could this be a typo, i.e. the dive being 24.3m? "The dive was 14.3m deep. Subsurface shows 24m."

wegi82, 2020-02-04:
[image]

As you can see in the screenshot on top there are the 24m. You will also see the dive itself was on the 14,3m. Also in the divelist the false 24m are shown.

atdotde, 2020-02-04:
Could you please, on desktop, export this dive to a Subsurface xml and post it here? You might have to change the extension to be allowed to upload it.

wegi82, 2020-02-10:
Hi! 
Back from a ice dive weekend attached the file.
[24m_failure.ssrf.pdf](https://github.com/Subsurface-divelog/subsurface/files/4183244/24m_failure.ssrf.pdf)

atdotde, 2020-02-11:
In fact, that ssrf file gives an explicit max depth of 24.079 m but the individual depth samples just go up to depth='14.497 m at time='13:54 min'. I don't know how that explicit max depth value got there, but you could manually delete it form the file and Subsurface will figure out the correct value from the samples.

A bit of background: Some dive computers store depth samples at a lower rate than actually measuring them. So it can happen that the maximum depth measured during a dive does not coincide with one of the stored samples. So in that case it makes sense to memorise the maxdepth explicitly and not figure it out from the (stored) samples.

Maybe the parsing of this maxdepth value is wrong for the i770r.

cdot, 2022-02-08:
I have also seen this happen with my i770r. I believe it is an artifact of the dive computer, and not of Subsurface.