subsurface/subsurface #2248: Ubuntu crash when clicking on flag

Saviq, 2019-08-19:
### Describe the issue:
- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Subsurface on Ubuntu crashes when clicking on a flag on the map. Originally reported [on the mailing list](https://groups.google.com/forum/#!category-topic/subsurface-divelog/linux-issues/IgoGXEpi4cM) for both AppImage and snap, I've confirmed it for the snap and got this, somewhat uninteresting, traceback:
```
#0  0x00007f855a2f5ec8 in QMetaObject::activate(QObject*, int, int, void**) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Core.so.5                                                                                    
#1  0x00007f855bdc8542 in QQuickMouseArea::clicked(QQuickMouseEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Quick.so.5                                                                                         
#2  0x00007f855bdc9daf in QQuickMouseArea::setPressed(Qt::MouseButton, bool, Qt::MouseEventSource) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Quick.so.5                                                            
#3  0x00007f855bdcae1a in QQuickMouseArea::mouseReleaseEvent(QMouseEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Quick.so.5                                                                                    
#4  0x00007f855bd40198 in QQuickItem::event(QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Quick.so.5                                                                                                          
#5  0x00007f855c2f682c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5                                                                             
#6  0x00007f855c2fe0f4 in QApplication::notify(QObject*, QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5                                                                                           
#7  0x00007f855a2c79a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Core.so.5                                                                                 
#8  0x00007f855bd528e0 in QQuickWindow::sendEvent(QQuickItem*, QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Quick.so.5                                                                                       
#9  0x00007f855bd56c48 in QQuickWindowPrivate::deliverMouseEvent(QQuickPointerMouseEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Quick.so.5                                                                    
#10 0x00007f855bd575d5 in QQuickWindowPrivate::deliverPointerEvent(QQuickPointerEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Quick.so.5                                                                       
#11 0x00007f855bd57df6 in QQuickWindowPrivate::handleMouseEvent(QMouseEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Quick.so.5                                                                                 
#12 0x00007f855b4fd375 in QWindow::event(QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5                                                                                                               
#13 0x00007f855bd596c5 in QQuickWindow::event(QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Quick.so.5                                                                                                        
#14 0x00007f855c2f682c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5                                                                             
#15 0x00007f855c2fe0f4 in QApplication::notify(QObject*, QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5                                                                                           
#16 0x00007f855a2c79a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Core.so.5                                                                                 
#17 0x00007f855c9f680a in QQuickWidget::mouseReleaseEvent(QMouseEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5QuickWidgets.so.5                                                                                
#18 0x00007f855c335038 in QWidget::event(QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5                                                                                                           
#19 0x00007f855c9f69eb in QQuickWidget::event(QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5QuickWidgets.so.5                                                                                                 
#20 0x00007f855c2f682c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5                                                                             
#21 0x00007f855c2fe64f in QApplication::notify(QObject*, QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5                                                                                           
#22 0x00007f855a2c79a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Core.so.5                                                                                 
#23 0x00007f855c2fd622 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5              
#24 0x00007f855c35014b in ?? () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007f855c3527ba in ?? () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007f855c2f682c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5                                                                             
#27 0x00007f855c2fe0f4 in QApplication::notify(QObject*, QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5                                                                                           
#28 0x00007f855a2c79a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Core.so.5                                                                                 
#29 0x00007f855b4f1523 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5                                                 
#30 0x00007f855b4f2ff5 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5                                   
#31 0x00007f855b4ca2eb in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5                                                 
#32 0x00007f85435991c0 in ?? () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#33 0x00007f8554314417 in g_main_context_dispatch () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0                                                                                                             
#34 0x00007f8554314650 in ?? () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#35 0x00007f85543146dc in g_main_context_iteration () from target:/snap/subsurface/1630/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0                                                                                                             [... cut ...]
### Operating system:
Ubuntu 19.04
Original report says Windows does not have this problem.

### Subsurface version:
4.9.0+

### Steps to reproduce:
1) open a dive list
2) click on a flag on the map

### Current behavior:
Crash.

### Expected behavior:
Selecting the dive in the dive list.

### Additional information:
By bisecting the snap revisions I have isolated the problem to this changeset: [ad4c8ca5f...32720a7c1](https://github.com/Subsurface-divelog/subsurface/compare/ad4c8ca5f...32720a7c1).

bstoeger, 2019-08-25:
@willemferguson reported the same problem (can't find the report though).I can reproduce this with the snap package, but not with my own build. That makes testing a bit hard, especially since the crash is deep in QML land.

Saviq, 2019-08-25:
Hey @bstoeger I wonder if it's related to the Qt version used? The snap has what's in Ubuntu 18.04, so `5.9.5+dfsg-0ubuntu2.1`.

bstoeger, 2019-08-25:
I'm using:
```
Subsurface v4.9.2-1-g4caee33c7eb5,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (ffbb47297539b873a1bc49d71ebd739639876b65)
built with Qt Version 5.12.2, runtime from Qt Version 5.12.2
built with libgit2 0.27.7
```

bstoeger, 2019-08-25:
When running the appimage under valgrind I hit these use-after free bugs:
```
==1030== Invalid read of size 8
==1030==    at 0x9350D71: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x9393CF6: QQmlNotifier::emitNotify(QQmlNotifierEndpoint*, void**) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x93336BB: QQmlData::signalEmitted(QAbstractDeclarativeData*, QObject*, int, void**) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x99FB411: QMetaObject::activate(QObject*, int, int, void**) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1030==    by 0x89A7481: QQuickMouseArea::clicked(QQuickMouseEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==    by 0x89A9380: QQuickMouseArea::setPressed(Qt::MouseButton, bool, Qt::MouseEventSource) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==    by 0x89A9CF0: QQuickMouseArea::mouseReleaseEvent(QMouseEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==    by 0x891F6EC: QQuickItem::event(QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==    by 0x823E88B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1030==    by 0x824596F: QApplication::notify(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1030==    by 0x99D1CA4: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1030==    by 0x8930152: QQuickWindow::sendEvent(QQuickItem*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==  Address 0x20d4e328 is 56 bytes inside a block of size 64 free'd
==1030==    at 0x4839E7B: operator delete(void*) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==1030==    by 0x9335110: QQmlData::destroyed(QObject*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x9A03552: QObject::~QObject() (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1030==    by 0x892197E: QQuickItem::~QQuickItem() (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==    by 0x893E615: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==    by 0x99F95C4: QObjectPrivate::deleteChildren() (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1030==    by 0x9A0388D: QObject::~QObject() (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1030==    by 0x892197E: QQuickItem::~QQuickItem() (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==    by 0x893E795: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==    by 0x99F95C4: QObjectPrivate::deleteChildren() (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1030==    by 0x9A0388D: QObject::~QObject() (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1030==    by 0x892197E: QQuickItem::~QQuickItem() (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==  Block was alloc'd at
==1030==    at 0x4838DBF: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==1030==    by 0x93CBD72: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x93CC9D9: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x93C9C56: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x93CA6AA: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x93CB69B: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x93CC9D9: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x93C9C56: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x93CA6AA: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x93CB69B: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x93CC9D9: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x93C9C56: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030== 
==1030== Invalid read of size 8
==1030==    at 0x9350D7A: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x9393CF6: QQmlNotifier::emitNotify(QQmlNotifierEndpoint*, void**) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x93336BB: QQmlData::signalEmitted(QAbstractDeclarativeData*, QObject*, int, void**) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1030==    by 0x99FB411: QMetaObject::activate(QObject*, int, int, void**) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1030==    by 0x89A7481: QQuickMouseArea::clicked(QQuickMouseEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==    by 0x89A9380: QQuickMouseArea::setPressed(Qt::MouseButton, bool, Qt::MouseEventSource) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==    by 0x89A9CF0: QQuickMouseArea::mouseReleaseEvent(QMouseEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==    by 0x891F6EC: QQuickItem::event(QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1030==    by 0x823E88B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1030==    by 0x824596F: QApplication::notify(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1030==    by 0x99D1CA4: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
[... cut ...]
==1030== ERROR SUMMARY: 1193989 errors from 835 contexts (suppressed: 8 from 4)
```
It appears that we free QtQuickItems that are still referenced somewhere in QML.

When running my own build under valgrind, these don't happen.

So it also appears that newer QML versions are more resilient against these bugs. Unfortunately without an older build environment, I don't even know where to start. @dirkhh: could you bisect this on Ubuntu 18.04? The offending commit is already narrowed down quite a lot.

bstoeger, 2019-08-25:
And here's a more complete backtrace (the depth of the stack traces is a bit ridiculous):
```
==1409== Invalid read of size 8
==1409==    at 0x9350D71: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1409==    by 0x9393CF6: QQmlNotifier::emitNotify(QQmlNotifierEndpoint*, void**) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1409==    by 0x93336BB: QQmlData::signalEmitted(QAbstractDeclarativeData*, QObject*, int, void**) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Qml.so.5)
==1409==    by 0x99FB411: QMetaObject::activate(QObject*, int, int, void**) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1409==    by 0x89A7481: QQuickMouseArea::clicked(QQuickMouseEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1409==    by 0x89A9380: QQuickMouseArea::setPressed(Qt::MouseButton, bool, Qt::MouseEventSource) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1409==    by 0x89A9CF0: QQuickMouseArea::mouseReleaseEvent(QMouseEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1409==    by 0x891F6EC: QQuickItem::event(QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1409==    by 0x823E88B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1409==    by 0x824596F: QApplication::notify(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1409==    by 0x99D1CA4: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1409==    by 0x8930152: QQuickWindow::sendEvent(QQuickItem*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1409==    by 0x8935F8A: QQuickWindowPrivate::deliverMouseEvent(QQuickPointerMouseEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1409==    by 0x89367D4: QQuickWindowPrivate::deliverPointerEvent(QQuickPointerEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1409==    by 0x893754D: QQuickWindowPrivate::handleMouseEvent(QMouseEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1409==    by 0x8C9AFBD: QWindow::event(QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Gui.so.5)
==1409==    by 0x8938A74: QQuickWindow::event(QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Quick.so.5)
==1409==    by 0x823E88B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1409==    by 0x824596F: QApplication::notify(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1409==    by 0x99D1CA4: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1409==    by 0x80DD561: QQuickWidget::mouseReleaseEvent(QMouseEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5QuickWidgets.so.5)
==1409==    by 0x827A5E6: QWidget::event(QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1409==    by 0x80DD752: QQuickWidget::event(QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5QuickWidgets.so.5)
==1409==    by 0x823E88B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1409==    by 0x8246028: QApplication::notify(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1409==    by 0x99D1CA4: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1409==    by 0x8244E88: QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1409==    by 0x8292D30: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1409==    by 0x8295292: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1409==    by 0x823E88B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1409==    by 0x824596F: QApplication::notify(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Widgets.so.5)
==1409==    by 0x99D1CA4: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1409==    by 0x8C8F47C: QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Gui.so.5)
==1409==    by 0x8C91034: QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Gui.so.5)
==1409==    by 0x8C6F4FA: QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Gui.so.5)
==1409==    by 0x124BB2CF: ??? (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5XcbQpa.so.5)
==1409==    by 0xC1FC9ED: g_main_context_dispatch (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6000.4)
==1409==    by 0xC1FCC87: ??? (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6000.4)
==1409==    by 0xC1FCD1B: g_main_context_iteration (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6000.4)
==1409==    by 0x9A2315B: QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1409==    by 0x99CFE0A: QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1409==    by 0x99D8593: QCoreApplication::exec() (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
[... cut ...]
==1409==    by 0x99FB952: QMetaObject::activate(QObject*, int, int, void**) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1409==    by 0x99FB952: QMetaObject::activate(QObject*, int, int, void**) (in /tmp/.mount_SubsurUVlfaQ/usr/lib/libQt5Core.so.5)
==1409==    by 0x67A5D1: MultiFilterSortModel::myInvalidate() (filtermodels.cpp:267)
==1409==    by 0x50D36A: MainWindow::refreshDisplay(bool) (mainwindow.cpp:413)
==1409==    by 0x5167EC: MainWindow::loadFiles(QStringList) (mainwindow.cpp:1710)
==1409==    by 0x4FF3A3: main (subsurface-desktop-main.cpp:109)
```

bstoeger, 2019-08-25:
Looking at the stack trace above:

In `MapLocationModel::reload()` we do a `beginResetModel()` / `endResetModel()` pair. The latter call frees a QQmlData object. That object is then accessed later from QML. At a first glance, this looks like a QML bug to me.

bstoeger, 2019-08-25:
If I replace the beginResetModel() / endResetModel() pair by distict *RemoveRows / *AddRows pairs, the thing still crashes:

```
==6824== Invalid read of size 8
==6824==    at 0x9350D71: ??? (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Qml.so.5)
==6824==    by 0x9393CF6: QQmlNotifier::emitNotify(QQmlNotifierEndpoint*, void**) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Qml.so.5)
==6824==    by 0x93336BB: QQmlData::signalEmitted(QAbstractDeclarativeData*, QObject*, int, void**) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Qml.so.5)
==6824==    by 0x99FB411: QMetaObject::activate(QObject*, int, int, void**) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Core.so.5)
==6824==    by 0x89A7481: QQuickMouseArea::clicked(QQuickMouseEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Quick.so.5)
==6824==    by 0x89A9380: QQuickMouseArea::setPressed(Qt::MouseButton, bool, Qt::MouseEventSource) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Quick.so.5)
==6824==    by 0x89A9CF0: QQuickMouseArea::mouseReleaseEvent(QMouseEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Quick.so.5)
==6824==    by 0x891F6EC: QQuickItem::event(QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Quick.so.5)
==6824==    by 0x823E88B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Widgets.so.5)
==6824==    by 0x824596F: QApplication::notify(QObject*, QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Widgets.so.5)
==6824==    by 0x99D1CA4: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Core.so.5)
==6824==    by 0x8930152: QQuickWindow::sendEvent(QQuickItem*, QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Quick.so.5)
==6824==    by 0x8935F8A: QQuickWindowPrivate::deliverMouseEvent(QQuickPointerMouseEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Quick.so.5)
==6824==    by 0x89367D4: QQuickWindowPrivate::deliverPointerEvent(QQuickPointerEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Quick.so.5)
==6824==    by 0x893754D: QQuickWindowPrivate::handleMouseEvent(QMouseEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Quick.so.5)
==6824==    by 0x8C9AFBD: QWindow::event(QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Gui.so.5)
==6824==    by 0x8938A74: QQuickWindow::event(QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Quick.so.5)
==6824==    by 0x823E88B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Widgets.so.5)
==6824==    by 0x824596F: QApplication::notify(QObject*, QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Widgets.so.5)
==6824==    by 0x99D1CA4: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Core.so.5)
==6824==    by 0x80DD561: QQuickWidget::mouseReleaseEvent(QMouseEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5QuickWidgets.so.5)
==6824==    by 0x827A5E6: QWidget::event(QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Widgets.so.5)
==6824==    by 0x80DD752: QQuickWidget::event(QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5QuickWidgets.so.5)
==6824==    by 0x823E88B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Widgets.so.5)
==6824==    by 0x8246028: QApplication::notify(QObject*, QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Widgets.so.5)
==6824==    by 0x99D1CA4: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Core.so.5)
==6824==    by 0x8244E88: QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Widgets.so.5)
==6824==    by 0x8292D30: ??? (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Widgets.so.5)
==6824==    by 0x8295292: ??? (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Widgets.so.5)
==6824==    by 0x823E88B: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Widgets.so.5)
==6824==    by 0x824596F: QApplication::notify(QObject*, QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Widgets.so.5)
==6824==    by 0x99D1CA4: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Core.so.5)
==6824==    by 0x8C8F47C: QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Gui.so.5)
==6824==    by 0x8C91034: QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Gui.so.5)
==6824==    by 0x8C6F4FA: QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Gui.so.5)
==6824==    by 0x124BB2CF: ??? (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5XcbQpa.so.5)
==6824==    by 0xC1FC9ED: g_main_context_dispatch (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6000.4)
==6824==    by 0xC1FCC87: ??? (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6000.4)
==6824==    by 0xC1FCD1B: g_main_context_iteration (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6000.4)
==6824==    by 0x9A2315B: QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Core.so.5)
==6824==    by 0x99CFE0A: QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (in /tmp/.mount_SubsurLBwnzu/usr/lib/libQt5Core.so.5)
[... cut ...]
==6824==    by 0x67A651: MultiFilterSortModel::myInvalidate() (filtermodels.cpp:267)
==6824==    by 0x50D36A: MainWindow::refreshDisplay(bool) (mainwindow.cpp:413)
==6824==    by 0x5167EC: MainWindow::loadFiles(QStringList) (mainwindow.cpp:1710)
==6824==    by 0x4FF3A3: main (subsurface-desktop-main.cpp:109)
```

Fatally reminds me of the crazy dive-selection bug where Qt messed up its internal state under very specific circumstances. Now this is combined with QML which makes debugging even more of a pain.

bstoeger, 2019-08-26:
@dirkhh: as long as nobody is able to debug or bisect this (I don't have access to Ubuntu 18.4, sorry), wouldn't it be possible to build the snap and appimage packages on newer Ubuntu? Kubuntu 19.4 works for me.

Saviq, 2019-08-27:
@bstoeger the only base OSs supported for snaps are currently 16.04 and 18.04, but [I'm working on](https://forum.snapcraft.io/t/bulding-qt-components-on-top-of-kde-frameworks/12930) getting newer Qt inside the snap.

bstoeger, 2019-08-27:
Sigh. Sadly, I have no machine on which I could install Ubuntu 18.04. And making  a PR for every bisection step to generate an appimage sounds not fun at all. :(

Saviq, 2019-08-27:
@bstoeger I will spend some time later today to pinpoint the exact breaking commit.

bstoeger, 2019-08-27:
Thanks - the good(?) news is that the SSD of my desktop-PC crashed. I will get a new one and install Ubuntu 18.04 on that. But it will take a few days.

Saviq, 2019-08-28:
Hey @bstoeger, the offending commit is 403206ca.

bstoeger, 2019-08-28:
Thanks for figuring that out. I hope this is not a different issue - because we had big problems in the selection part of the code.

bstoeger, 2019-08-28:
> ever heard of virtual machines?

Yes, but I never used one, because up to now it wasn't necessary. And I'm not sure spending the energy and time to learn using it would be well spent at the moment. I wouldn't even know where to start looking.

bstoeger, 2019-08-30:
FYI: I got my new desktop SSD and installed Ubuntu 18.4. I believe this to be a Qt bug that has been fixed, but I didn't find a bug report. Under certain circumstances resetting the model destroys Qt's internal state. The same thing we had for the selection and the QSortFilterProxy model.

I'm working on a fix that involves not resetting the whole model with *ResetModel(), but simply doing a dataChanged() when the selection changes.

dirkhh, 2019-08-30:
Fascinating that this would be a silently fixed Qt bug. @thiagomacieira any idea if that's a thing?
I like the idea of not handling selection changes via `dataChanged()` - should 4.9.3 wait for this?

bstoeger, 2019-08-30:
Well, matter of fact Qt 5.9 crashes owing to accessing data that it freed itself. Qt 5.12 doesn't do that.

Yes, maybe wait, because an AppImage that crashes when clicking on the map is embarrassing. But I'm not sure that I'll get it done because I'm now heading off for dinner and will not have access to the Ubuntu 18.4 machine on weekend. Though I know what to do to keep it from crashing: don't reset the model.

thiagomacieira, 2019-08-31:
Doesn't the appimage come with the latest Qt?

dirkhh, 2019-08-31:
No it picks up the QT version of the distro it's built on, which tends to be older.

bstoeger, 2019-08-31:
> Doesn't the appimage come with the latest Qt?

That doesn't help us for people compiling on old Qt (we're supposed to support at least Qt 5.5) and vendor-supplied packages.

dirkhh, 2019-08-31:
Distro specific binaries will be using the distro Qt. So Ubuntu 18.04 native binaries will crash. That sucks. But so be it. If we can solve the AppImage problem (which I'm working on by trying to get a Trusty with Qt 5.12.4 to work), then we should have a solution that we can point anyone on Linux to.

bstoeger, 2019-08-31:
Or provide a build which includes #2273? Risky, but better than crashing.
Or, as you suggested, temporarily disable maps for Qt < 5.x until #2273 is ready?

dirkhh, 2019-08-31:
I can still make #2273 crash...
I'm more leaning towards disabling maps for older Qt

bstoeger, 2019-08-31:
Ok, I will have to check that out. Perhaps it's a different crash that is introduced by #2273. :-P

dirkhh, 2019-09-01:
Still trying to get a viable build with Qt 5.12 on Trusty. QtWebKit is failing. And unfortunately that's a build that is incredibly painful to debug - every new attempt takes >1h

PR #2273 (2019-09-06): Fix dive site selection on Qt 5.9

### Describe the pull request:

- [x] Bug fix
- [x] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

This is an attempt at fixing the crashes when clicking on a dive site with Qt 5.9. The thing is to not reset the model, which apparently trashes Qt's internal state, but posting a dataChanged() signal.

Notes:
1) I detest the code that I have written here - but let's be pragmatic for now.
2) Currently this makes editing of dive sites impos [... cut]
closed 2019-09-06 by dirkhh