subsurface/subsurface #2251: crash with "Save dive data as subtitle"

fmatray, 2019-08-19:
### Describe the issue:
- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Crash (SIGABORT) with "Save dive data as subtitle"

### Operating system:
Mac OS 10.14.5 French (Up to date)

### Subsurface version:
Subsurface 4.9.1

### Steps to reproduce:
- Select a dive with media
- Right Click on "Save dive data as subtitle"
-> crash with sigabort

### Current behavior:
Crash

### Expected behavior:
Having the subtitles

### Additional information:
prometheus:MacOS $ ./Subsurface -v -v -v 
[...]
item  QDeclarativeGeoMapQuickItem(0x7fa69f583370, parent=0x0, geometry=152.828,44.6996 32x32)  still referenced
Subsurface(16750,0x11b3fc5c0) malloc: *** error for object 0x7ffee2401060: pointer being freed was not allocated
Subsurface(16750,0x11b3fc5c0) malloc: *** set a breakpoint in malloc_error_break to debug
Abort trap: 6
prometheus:MacOS $ 

### Mentions:

atdotde, 2019-08-19:
Could you share the dive file (via export) with me so I can have a look? helling@atdotde.de

bstoeger, 2019-08-25:
Can reproduce - will look into it.

commit 4caee33c (2019-08-25): Profile: properly initialize plot_info structures

The create_plot_info_new() function releases old plot data. This
can only work if the plot_info structure was initialized previously.
The ProfileWidget2 did that by a memset, but other parts of the code
did not.

Therefore, introduce a init_plot_info() function and call that when
generating a plot_info struct. Constructors would make this so much
easier - but since this is called from C, we can't use them.

Fixes #2251
commit bf293d4b (2019-08-25): Profile: properly initialize plot_info structures

The create_plot_info_new() function releases old plot data. This
can only work if the plot_info structure was initialized previously.
The ProfileWidget2 did that by a memset, but other parts of the code
did not.

Therefore, introduce a init_plot_info() function and call that when
generating a plot_info struct. Constructors would make this so much
easier - but since this is called from C, we can't use them.

Fixes #2251
commit 25b30da2 (2019-08-28): Profile: properly initialize plot_info structures

The create_plot_info_new() function releases old plot data. This
can only work if the plot_info structure was initialized previously.
The ProfileWidget2 did that by a memset, but other parts of the code
did not.

Therefore, introduce a init_plot_info() function and call that when
generating a plot_info struct. Constructors would make this so much
easier - but since this is called from C, we can't use them.

Fixes #2251