subsurface/subsurface #2253: Crash when copy / PASTE dive information

sfuchs79, 2019-08-24:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Subsurface crashes when copy/PASTE dive info from one dive to another.

### Operating system:

Windows 10

### Subsurface version:

Own MXE build 4.9.1-59 (based on current master with a few unrelated changes)

### Steps to reproduce:

- Select one dive
- Press CTRL-C
- Select items to copy (e.g. weight, tank,...)
- Select another dive
- Press CTRL-V
--> Crash

No trace output :-(
Some limited info from Dr.MinGW:
subsurface.exe caused an Access Violation at location 0000000000536A94 in module subsurface.exe Reading from location 0000000022353238.

Registers:
eax=22353234 ebx=00000000 ecx=00000000 edx=0baad9e8 esi=1ccdf6cc edi=1ccdf6cc
eip=00536a94 esp=057fd670 ebp=1ccdf1f8 iopl=0         nv up ei pl nz na po nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00210206

AddrPC   Params
00536A94 000004E0 00000200 1D8C8AF8  subsurface.exe!clear_weightsystem_table  [/home/stefan/Entwicklung/Subsurface/subsurface/core/equipment.c @ 24]
6FEF8EBA 00000000 00000000 00000000  libstdc++-6.dll!operator new

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

sfuchs79, 2019-08-24:
Mini bisect: 4.9.0 doesn't crash.

bstoeger, 2019-08-24:
Can reproduce - will look into it.

commit e6334aa3 (2019-08-24): Desktop: fix crash on copy & paste

The constructor of PasteState was clearing an uninitialized
weightsystem-table. Very silly. Initialize it instead.

Fixes #2253

Reported-by: Stefan Fuchs <sfuchs@gmx.de>
PR #2254 (2019-08-24): Fix copy & paste crash

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Fix silly bug in PasteState constructor.
### Changes made:

1) Initialize weightsystem_table, don't free it.

### Related issues:

Fixes #2253
### Additional information:

### Release note:

Done.
### Documentation change:
commit 97380d6a (2019-08-25): Desktop: fix crash on copy & paste

The constructor of PasteState was clearing an uninitialized
weightsystem-table. Very silly. Initialize it instead.

Fixes #2253

Reported-by: Stefan Fuchs <sfuchs@gmx.de>
closed 2019-08-24 by dirkhh