subsurface/subsurface #2261: Gas consumption in planner notes not displayed corretly - put_format() bug?!

sfuchs79, 2019-08-25:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Gas consumption in planner notes not displayed correctly. Part of the string missing:
[image]

I did some debugging and I have some doubts that I found a bug in put_format()
It may be that the function is processing some strings incorrectly.

I added this code to plannernotes.c:
`/* Gas consumption: Now finally print all strings to output */
printf("----- This goes to buf using put_format():\n");
printf("%s%s%s<br>\n\n\n", temp, warning, mingas);
put_format(&buf, "%s%s%s<br>", temp, warning, mingas);
printf("----- This is the buf after:\n");
printf("%s\n\n\n", buf.buffer);
free(temp);`

And I see in the output the part of one string is not transferred correctly to the buffer:
[debug_planner_notes_buffer.txt](https://github.com/Subsurface-divelog/subsurface/files/3538527/debug_planner_notes_buffer.txt)

### Operating system:

Windows 10

### Subsurface version:

Latest master

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@bstoeger

bstoeger, 2019-08-25:
Well, at first glance I don't see anything wrong with `put_format()`, though it's old-school C-code and I have very little experience with that.

What I see though is horribly non-balanced `<div>` tags...

bstoeger, 2019-08-25:
@atdotde: I can hardly debug this, as I don't know what the expected output is. But clearly the div-tags are wrong - there is a single opening tag before the gases, but one closing tag per gas.

sfuchs79, 2019-08-26:
Yes, I recognised the non-balanced tags as well and fixed them but they are not the cause.
I will try s.th. else later but to be honest I have no glue what is going wrong.

bstoeger, 2019-08-26:
You could add the debug-printfs directly in the put_vformat function. Interesting points would be: before and after the vsnprintf and the return of vsnprintf.

I think it should be OK, because vsnprintf is always 0-terminating.

atdotde, 2019-08-26:
Could you please share a planned dive (or full planner screen shot) that demonstrates the problem as I cannot reproduce it.

sfuchs79, 2019-08-26:
New finding, see #2263 but not really perfect but more dissatisfying because I don't understand whats happening with put_format(). Can do more debugging but only later...

bstoeger, 2019-08-26:
> Can do more debugging but only later...

I would really prefer that over using the *_loc version. Something is seriously wrong here.

sfuchs79, 2019-08-27:
I agree with @bstoeger .
Let's try to debug more. As a first step, here is a planned dive and the settings I can use to reproduce the issue:

[Planner_notes_bug.zip](https://github.com/Subsurface-divelog/subsurface/files/3547068/Planner_notes_bug.zip)

[image]

sfuchs79, 2019-08-27:
Ähhmmm, question: What does this value do:
https://github.com/Subsurface-divelog/subsurface/blame/master/core/membuffer.c#L91

Because if I increase this to e.g. 1024 the issue with put_format() also disappears.

glance-, 2019-08-27:
I looked at plannernotes.c html and saw a lot of unmatched tags , and did some quick hacking to produce something which a validator accepts.
[plannernotes.txt](https://github.com/Subsurface-divelog/subsurface/files/3548047/plannernotes.txt)

It wastes a couple of bytes by using \n and \t, but that makes the html way more readable.

bstoeger, 2019-08-28:
This is what I get when switching to German: 
[image]
I reckon that is correct?

> What does this value do:

This is the initial buffer size. If the vnsprintf call doesn't have enough room, it will reallocated the buffer.

Could you please print out the return of the vnsprintf, the len and the resulting buffer.
Something like:
```
printf("b->len: %u, b->alloc: %u len: %d b->buffer: %s\n", b->len, b->alloc, len, b->buffer);
```

Thanks.

sfuchs79, 2019-09-22:
Fixed by #2266

closed 2019-09-22 by sfuchs79