subsurface/subsurface #2264: Feature of identifying old planner notes in dive notes defect

sfuchs79, 2019-08-27:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

There was once a feature which automatically identified old planner notes in planned dive notes.
It was/is implemented here:
https://github.com/Subsurface-divelog/subsurface/blob/master/qt-models/diveplannermodel.cpp#L1130

But this feature needs the global variable "disclaimer" holding the translated disclaimer text.

And filling this global variable was removed here:
https://github.com/Subsurface-divelog/subsurface/commit/a4834e198d55d50a44d6f80e2137777b64652ce7#diff-0df10959bbcbe5342528360ce6f541a5L116

We please should have a look together how to make this work again :-)

### Operating system:

### Subsurface version:

Current master

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@atdotde @bstoeger

bstoeger, 2019-09-01:
Do I understand correctly that it is supposed to remove the old disclaimer when replanning a dive?

But who did that even work if it depends on the type of the previously planned dive? That appears extremely brittle to me.

Do you have any suggestion how to fix that?

sfuchs79, 2019-09-22:
Fixed by #2274

closed 2019-09-22 by sfuchs79