subsurface/subsurface #2269: Linker Errors with LibGit2 & LibCurl4

keithgrimes, 2019-08-29:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [X] Discussion request

### Issue long description:
Have been compiling Subsurface on my RaspberryPi3 for sometime now with no issues (Raspbian Stretch/Debian 9) due to it being an arm processor. 
I have now moved to a new RaspberryPi4 running Raspbian Buster / Debian 10 and get a linker error against LibCurl4 with CURL_OPENSSL_3 not found. 

### Operating system:
 Raspbian Buster / Debian 10
 Raspberry Pi 4

### Subsurface version:
 4.9.4 (Compiled Locally)
 Yes (latest build)
 Self Compile
 Latest (not using one)
LibGit2 using 15e119375018fba121cf58e02a9f17fe22df0df8 (v26.0)

### Steps to reproduce:

Clean installation of Raspbian Buster.
Compilation on a Pi4 Model B. Running Buster OS (Debian 10 port). No special changes 

### Current behavior:

Following Linker Errors in several places

/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../libgit2.so: undefined reference to `curl_easy_init@CURL_OPENSSL_3'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../libgit2.so: undefined reference to `curl_easy_getinfo@CURL_OPENSSL_3'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../libgit2.so: undefined reference to `curl_easy_perform@CURL_OPENSSL_3'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../libgit2.so: undefined reference to `curl_easy_send@CURL_OPENSSL_3'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../libgit2.so: undefined reference to `curl_easy_setopt@CURL_OPENSSL_3'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../libgit2.so: undefined reference to `curl_easy_cleanup@CURL_OPENSSL_3'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../libgit2.so: undefined reference to `curl_easy_recv@CURL_OPENSSL_3'

Note Debian Buster ships with LibCurl4 (not LibCurl3). LibCurl3 is not available as an installation option.

### Expected behavior:
No Error message (link successfully finds the references). 

### Additional information:
Build Log attached. 
I have investigated and believe the original compilation (under Debian Stretch/Debian 9) used LibCurl3 which implements OPENSSL 1.0. However, Debian 10 (and hence Raspian Buster) installs LibCurl4 which implements OPENSSL 1.1. These are not mutually compatible and hence it does not find the reference. LibCurl3 is not available as an installation package as it conflicts with LibCurl4. 

Can anyone suggest a route forward?

### Mentions:

[build.log](https://github.com/Subsurface-divelog/subsurface/files/3557368/build.log)

buxtonpaul, 2019-09-24:
Based on this https://github.com/curl/curl/issues/2433
It looks like the libcurl3/4 compatability is something debian have introduced!
Suggest maybe build curl library from source...
https://curl.haxx.se/libcurl/abi.html

keithgrimes, 2019-09-24:
> Based on this curl/curl#2433

Hi Paul. 
Thanks for the suggestion, I may give it a go. In the interim I have the software running in a Docker container which seems to suffice.

Thanks
Keith

closed 2019-10-28 by dirkhh