subsurface/subsurface #2272: Newly imported dives aren't numbered

kittykat, 2019-08-30:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When importing dives from a Suunto D4i, the newly imported dives are all numbered "0" instead of continuing from the last dive as used to happen in older versions

### Operating system:
Linux

### Subsurface version:
4.9.2-5-g3c4fd5d5999d

### Steps to reproduce:
1) Import dives from a computer using the "import from dive computer" dialog

### Current behavior:
Newly imported dives are always numbered "0"

### Expected behavior:
Dive numbering continues from last imported dive

kittykat, 2019-09-02:
I have my open water dives numbered chronologically starting from 1. My pool dives are numbered chronologically starting from 20000 and upwards (as I can't exclude them from the numbering any other way that I could see). My latest dive is an open water dive with the correct number for that dive (as are 10 or so before it).  In previous versions, the numbering used to continue from the latest dive, which was convenient.

dirkhh, 2019-10-28:
They only continue on if the last dive also has the largest number. Your numbering strategy is just something we don't support.

closed 2019-10-28 by dirkhh