subsurface/subsurface #2275: Selected dive sites export is confusing

fmatray, 2019-09-03:
#1637 # Describe the issue:
- [ ] Bug
- [X] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I select several dive sites in "dive sites" and try to export (File -> Export -> Subsurface dive sites XML) selected dives. The first dive site is exported only.

As I selected several dive sites, I expect to have those exported.
To make it work, I have to select dives after dive sites selection.

This is a bit confusing. When I select several dive sites, I expect to have related dives selected also.

### Operating system:
MacOS Mojave

### Subsurface version:
Subsurface 4.9..1

### Steps to reproduce:
1) Select several dive sites
2) File -> Export
3) Chose "Subsurface dive sites XML" and "selected dives"
4) Export

### Current behavior:
One dive site exported

### Expected behavior:
Selected dive sites exported

### Additional information:
[image]

[sainte maxime.xml.zip](https://github.com/Subsurface-divelog/subsurface/files/3571783/sainte.maxime.xml.zip)

### Mentions:

bstoeger, 2019-09-05:
Yes, that's silly. I haven't looked into this, but I suspect that the problem is that the export works from normal-mode and from dive-site-mode.

@dirkhh: What do you think: should we change the export logic if on the dive-site-tab? Or perhaps select all dives while the dive-site-tab is active?

dirkhh, 2019-09-05:
When opened from the dive site screen it should be "export selected dive sites", right?

bstoeger, 2019-09-05:
That appears logical, yes. Since the dialog is modal, we can change it depending on where it was called.

djunkins, 2019-09-05:
Sorry — I’ve been distracted with other things for the last several weeks. I can look at this early next week if you want.

-Doug

bstoeger, 2019-09-21:
Fix attempt in #2293, though the UI is not clear about what is going on. This could still need some work.

PR #2293 (2019-09-21): Desktop: In dive site mode export selected dive sites

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [x] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

This fixes an unintuitivness of the UI: To export only the selected dive sites, the user had to select all dives. Instead, when in dive site mode, consider the selected dive sites instead of the selected dives. 
### Changes made:

1) Numerous small cleanups.
2) Special case the dive site mode on dive site export.
### Re [... cut]
commit 4f0582ea (2019-09-21): Export: special case dive site mode when exporting dive sites

In dive site mode, export selected dive sites, not dive sites
of selected dives.

Fixes #2275.
closed 2019-09-21 by dirkhh