subsurface/subsurface #2291: Mares Quad bluetooth transfer fail

tiberioalunni, 2019-09-16:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

Windows 10

### Subsurface version:

4.9.3 official release

### Steps to reproduce:

Try to import dive from Mares Quad

### Current behavior:

Subsurface seems to connect to device correctly, during data transfer an error appear:

[image]

### Expected behavior:

Show dive log list to download

### Additional information:

libdivecompurer.log

> Subsurface: v4.9.3, built with libdivecomputer v0.7.0-devel-Subsurface-NG (ce6d9896a79afaa82641132e338f8744714c8593)
INFO: Open: transport=32
INFO: Configure: baudrate=115200, databits=8, parity=2, stopbits=0, flowcontrol=0
INFO: Timeout: value=1000
INFO: DTR: value=0
INFO: RTS: value=0
INFO: Purge: direction=3
INFO: Write: size=2, data=C267
INFO: Read: size=0, data=
ERROR: Failed to receive the answer. [in /win/subsurface/libdivecomputer/src/mares_iconhd.c:270 (mares_iconhd_packet)]
INFO: Purge: direction=1
INFO: Write: size=2, data=C267
INFO: Read: size=0, data=
ERROR: Failed to receive the answer. [in /win/subsurface/libdivecomputer/src/mares_iconhd.c:270 (mares_iconhd_packet)]
INFO: Purge: direction=1
INFO: Write: size=2, data=C267
INFO: Read: size=20, data=FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
ERROR: Unexpected answer byte. [in /win/subsurface/libdivecomputer/src/mares_iconhd.c:276 (mares_iconhd_packet)]
INFO: Purge: direction=1
INFO: Write: size=2, data=C267
INFO: Read: size=20, data=FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
ERROR: Unexpected answer byte. [in /win/subsurface/libdivecomputer/src/mares_iconhd.c:276 (mares_iconhd_packet)]
INFO: Purge: direction=1
INFO: Write: size=2, data=C267
INFO: Read: size=9, data=FFFFFFFFFFFFFFFFFF
ERROR: Unexpected answer byte. [in /win/subsurface/libdivecomputer/src/mares_iconhd.c:276 (mares_iconhd_packet)]

### Mentions:

dirkhh, 2019-09-16:
This actually looks like it's trying to connect over a serial line - it's missing a lot of the tell-tale BLE conversation information I would expect.

tiberioalunni, 2019-09-20:
@dirkhh should I try some different connection configuration?

dirkhh, 2019-10-28:
I'm not sure why it does what it does. One thing to try is to unpair the dive computer from Windows, then pair again and try again.
The BLE stack on Windows is extremely flaky - we frequently run into issues that just go away when trying again.

tiberioalunni, 2019-12-02:
@dirkhh tried many pair/unpair, but the problem persist.

LECbg, 2020-02-21:
Hello, I'm getting a similar problem:

**Operating system**: MacOS Catalina (10.15.3)
**Subsurface version**: 4.9.3 official release

**Steps to reproduce**:
Try to import dive from Mares Puck Pro + with Mares Dive Link.
**Current behavior**:
Subsurface seems to connect to device correctly, Dive Link even blinks green (as it should during a data transfer), but an error appears:
[image]
Sorry, my OS is in Spanish, it basically tells the user to search the logs, and here's what's in subsurface.log:

> Subsurface: v4.9.3, built with libdivecomputer v0.7.0-devel-Subsurface-NG (ce6d9896a79afaa82641132e338f8744714c8593)
> INFO: Open: transport=32
> INFO: Configure: baudrate=115200, databits=8, parity=2, stopbits=0, flowcontrol=0
> INFO: Timeout: value=1000
> INFO: DTR: value=0
> INFO: RTS: value=0
> INFO: Purge: direction=3
> INFO: Write: size=2, data=C267
> INFO: Read: size=0, data=
> ERROR: Failed to receive the answer. [in ../../src/mares_iconhd.c:270 (mares_iconhd_packet)]
> INFO: Purge: direction=1
> INFO: Write: size=2, data=C267
> INFO: Read: size=0, data=
> ERROR: Failed to receive the answer. [in ../../src/mares_iconhd.c:270 (mares_iconhd_packet)]
> INFO: Purge: direction=1
> INFO: Write: size=2, data=C267
> INFO: Read: size=0, data=
> ERROR: Failed to receive the answer. [in ../../src/mares_iconhd.c:270 (mares_iconhd_packet)]
> INFO: Purge: direction=1
> INFO: Write: size=2, data=C267
> INFO: Read: size=0, data=
> ERROR: Failed to receive the answer. [in ../../src/mares_iconhd.c:270 (mares_iconhd_packet)]
> INFO: Purge: direction=1
> INFO: Write: size=2, data=C267
> INFO: Read: size=0, data=
> ERROR: Failed to receive the answer. [in ../../src/mares_iconhd.c:270 (mares_iconhd_packet)]

It looks exactly like the first part from [the log above](https://github.com/Subsurface-divelog/subsurface/issues/2291#issue-493912137) and that's why I posted this here.

Any guess on what's going on here? Are there any other log files I can look for?