subsurface/subsurface #2292: Not Downloading new dives

chartb, 2019-09-21:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Not downloading the latest dives via Bluetooth.  The connection is made as Subsurface displays the serial number of the dive computer but it states no new dives since 30th June the last time I was diving.  The manufacturers software has downloaded the dives from this month successfully.
### Operating system:

Android 8.1.0

Samsung Galaxy J5

### Subsurface version:

2.2.0(4.9.1.18)

Official Release

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

---------- subsurface.log ----------
"0.008: Failed to open logfile /storage/emulated/0/subsurface.log at Fri Sep 20 22:56:46 2019 error: Permission denied"
"0.008: Failed to open logfile /storage/emulated/0/Documents/subsurface.log at Fri Sep 20 22:56:46 2019 error: Permission denied"
"0.009: Successfully opened logfile /storage/emulated/0/Android/data/org.subsurfacedivelog.mobile/files/Documents/subsurface.log at Fri Sep 20 22:56:46 2019"
"0.011: Starting Subsurface-mobile:2.2.0(4.9.1.18):Android  (8.1):arm:en-GB"
"0.011: built with libdivecomputer v0.7.0-devel-Subsurface-NG (ffbb47297539b873a1bc49d71ebd739639876b65)"
"0.011: built with Qt Version 5.12.4, runtime from Qt Version 5.12.4"
"0.012: built with libgit2 0.26.0"
BTDiscoveryReDiscover: localBtDevice.isValid() true
"BTDiscoveryReDiscover: localDevice Galaxy J5 is powered on, starting discovery"
paired BLE device type 2 with address "LE:50:22:61:82:AE:2D"
paired BT classic device type 1 with address "12:34:56:72:96:36"
paired BT classic device type 1 with address "E4:03:28:67:09:54"
paired BT classic device type 1 with address "90:C4:BA:A7:AE:3D"
paired BT classic device type 1 with address "10:C6:CF:4C:F4:D5"
paired BLE device type 2 with address "LE:00:13:43:69:B7:8E"
Found new device: "TomTom" "LE:50:22:61:82:AE:2D"
Not recognized as dive computer
Found new device: "OBDII" "12:34:56:72:96:36"
Not recognized as dive computer
Found new device: "TomTom" "E4:03:28:67:09:54"
Not recognized as dive computer
Found new device: "RFRF711" "90:C4:BA:A7:AE:3D"
Not recognized as dive computer
Found new device: "Car Phone #F4D5" "10:C6:CF:4C:F4:5D"
Not recognized as dive computer
Found new device: "Aladin Sport Matrix" "LE:00:13:43:69:B7:8E"
"this could be a Scubapro Aladin Sport Matrix"
Paired = "TomTom" "LE:50:22:61:82:AE:2D"
Paired = "OBDII" "12:34:56:72:96:36"
Paired = "TomTom" "E4:03:28:67:09:54"
Paired = "RFRF711" "90:C4:BA:A7:AE:3D"
Paired = "Car Phone #F4D5" "10:C6:CF:4C:F4:5D"
Paired = "Aladin" "LE:00:13:43:69:B7:8E"
"Created position source android"
"0.048: Created position source android"
"Set GPS service update interval to 300 s"
"0.049: Set GPS service update interval to 300 s"
"0.049: location service is not available"
qrc:/org/kde/kirigami/Page.qml:276:5: QML Loader: Binding loop detected for property "active"
[... 2212 lines cut ...]
INFO: Read: size=20, data=130211700475027019750F047E03C613026E0B76
DEBUG: rcv: size=19, data=0211700475027019750F047E03C613026E0B76
INFO: Read: size=20, data=137E047606037E04720502C17F037F0273C10A01
DEBUG: rcv: size=19, data=7E047606037E04720502C17F037F0273C10A01
INFO: Read: size=20, data=13C104797C75126F147C7C7E78737D7B75717877
DEBUG: rcv: size=19, data=C104797C75126F147C7C7E78737D7B75717877
INFO: Read: size=20, data=137B776B71706C72EA59E27DE07EC1067B0C0D7C
DEBUG: rcv: size=19, data=7B776B71706C72EA59E27DE07EC1067B0C0D7C
INFO: Read: size=20, data=130701027C7701087F037E07027501017C7E017D
DEBUG: rcv: size=19, data=0701027C7701087F037E07027501017C7E017D
INFO: Read: size=20, data=137A7F077B037A7C067E7EC17D7E7FB107017E03
DEBUG: rcv: size=19, data=7A7F077B037A7C067E7EC17D7E7FB107017E03
INFO: Read: size=20, data=137F02726FE266E0686F776F707A70747774E16C
DEBUG: rcv: size=19, data=7F02726FE266E0686F776F707A70747774E16C
INFO: Read: size=20, data=13E07478780875C1077D7376047D7F7F037F146F
DEBUG: rcv: size=19, data=E07478780875C1077D7376047D7F7F037F146F
INFO: Read: size=20, data=137F7DC2017FC2047E0876C47FC101C17FC10102
DEBUG: rcv: size=19, data=7F7DC2017FC2047E0876C47FC101C17FC10102
INFO: Read: size=20, data=137F7FC27FC1017D02C67C04C37F027FC901FB10
DEBUG: rcv: size=19, data=7F7FC27FC1017D02C67C04C37F027FC901FB10
INFO: Read: size=16, data=0F201500000000000000780500000000
DEBUG: rcv: size=15, data=201500000000000000780500000000
---------- finish ----------

### Mentions:

mturkia, 2019-09-21:
The log file shows that there are 7 new dives in September. Just to make sure, did you try downloading without checking the "Save libdivecomputer dump file"? If this option is selected, we do not import any dives, but save full memory dump of the DC into a file instead. So when you are downloading new dives, you should make sure this option is not selected.

chartb, 2019-09-21:
Hi,

I have looked but cannot see how to change the option "Save libdivecomputer dump file". I have tried "force download all dives".

Are there any options I can select in the menus that will allow me to change this option?

Many thanks

Barry

Sent from ProtonMail mobile

-------- Original Message --------

mturkia, 2019-09-21:
I missed the fact that you are on mobile. There is no such thing as libdivecomputer dump file, so in that sense you should be good to go.

The following lines in the log seem peculiar to me:
 _turning off further BT debug output
"66.393: Dive 1: Sun 15 Sep 2019 12:43"
"66.403: Dive 2: Sun 15 Sep 2019 8:38"
"66.412: Dive 3: Sat 14 Sep 2019 12:30"
"66.419: Dive 4: Sat 14 Sep 2019 9:10"
"66.424: Dive 5: Sun 8 Sep 2019 11:01"
"66.431: Dive 6: Sat 7 Sep 2019 14:45"
"66.434: Dive 7: Sat 7 Sep 2019 10:36"
"66.438: Dive 8: Sun 30 Jun 2019 14:02"
"66.441: Already downloaded dive at Sun 30 Jun 2019 14:02"
Deleting BLE object
Finishing download thread: 7 dives downloaded
no new dives downloaded
"66.452: DCDownloadThread finished"_

It clearly states that 7 dives downloaded - this is logged from core (core/downloadfromdcthread.cpp). The next line claiming no new dives downloaded is in QML (mobile-widgets/qml/DownloadFromDiveComputer.qml). For some reason dcImportModel.rowCount is zero, even though some dives were downloaded. I hope @bstoeger or @dirkhh have an idea, what is going on.

bstoeger, 2019-09-22:
Does this include #2219? Then this is probably the crash described in #2221: The passed in diveTable is NULL. Which it can't be.

I still haven't setup a working mobile-on-desktop. :( My current laptop doesn't support Bluetooth (Apparently I would have to do some kernel module compilation). I'll try to work on that.

dirkhh, 2019-09-22:
I just pushed a new update to Subsurface-mobile that shouldn't encounter this problem. Would you please try that @chartb ?

bstoeger, 2019-09-22:
> I just pushed a new update to Subsurface-mobile that shouldn't encounter this problem.

For my understanding: Which commit (probably) solves this problem?

I'm currently trying to remove the passing of table-pointers through QML, because obviously there are very weird things happening there! Like: we pass a pointer that we know is not null and it is null in the called function.

dirkhh, 2019-09-22:
Maybe I misunderstood your comment above that this might be caused by #2219 (which the current version no longer includes)

bstoeger, 2019-09-22:
Oh, I was not properly expressing myself then. The way I understand: #2219 makes a crash condition (diveTable is null) into a won't crash but dives are not imported.

If this is correct, then this will now crash, but at least we know where the problem is.

This part of #2219 cannot cause the problem - if diveTable is null something crazy happened before. :(

bstoeger, 2019-09-22:
Ouch - it might not be the direct cause of this bug, but it just occurred to me that what we're doing is fundamentally wrong:

Note how in DiveImportModel::repopulate() we set a pointer to the dive-table of the download thread.
When the thread starts downloading, it resets the dive table, but the model still believes that it has the previous number of dives.
Accessing the model while the download is in progress will now return dangling references!

In that light it's no wonder that we experience crashes and data corruption when downloading.

What I think we should do is: Keep the dive- and site-lists in the model, not the thread. The thread records into his own list. When finished, the list is *moved* from the thread to the model. Nobody ever accesses the list of the thread (except the thread itself).

chartb, 2019-09-22:
I have just installed the updated app from gplay and it started fine signed in to the cloud storage selected to download new dives and at the end it crashed. When I restarted it opens in to the dive list and the menus have disappeared. There are no new dives recorded.

Sent from ProtonMail mobile

-------- Original Message --------

bstoeger, 2019-09-22:
Yes, this confirms my suspicion that it's the #2221 crash, which is hidden by #2219. @dirkhh: better revert to the older version that at least doesn't crash, no?

dirkhh, 2019-09-22:
I'd actually prefer to figure out WHY this crashes.
@chartb, are you by any chance a software developer? Or comfortable with the Android development tools?
Google does make crash information available to the developers - but completely anonymized and it's often hard to figure out which crash matches which bug report. But if you happen to know how to use `adb` then it's easy for you to give us the crash stack trace. No big deal if the answer is "no", it's just worth asking...

chartb, 2019-09-23:
It's not something that I have done, but will look into it.

Sent from ProtonMail mobile

-------- Original Message --------

bstoeger, 2019-10-03:
What's the status of this with #2305 merged?

dirkhh, 2019-10-03:
I am hopeful that this should be fixed. Would appreciate a report back with the latest Subsurface-mobile BETA from Google Play (come to think of it - I will likely move that build to production now... we had plenty of positive feedback)

chartb, 2019-10-10:
App downloaded and it works, as all new dives are recorded.

Sent from ProtonMail mobile

-------- Original Message --------

bstoeger, 2019-10-16:
I interpret that as fixed and close.

closed 2019-10-16 by bstoeger