subsurface/subsurface #2294: Subsurface 4.9.3 - No more Dive suit field on the screen

msd3287, 2019-09-22:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I just upgrade Subsurface from 4.8.6 to 4.9.3.
When I open my log book I can't see anymore field "Drive suit" (Combinaison , in french).
I tried to install 4.9.1 in replacement and, than, the field is there visible (see  print screens below)
1) 4.9.1 with field "Combinaison"
[image]

2) 4.9.3 without field "Combinaison"
[image]

- Another point: 
1) the old filter screen (4.8.x) was better and more user friendly than the actual one (4.9.x) as previously it was possible to chose within the existing labels what to filter but, now (4.9), you have to fully write what you want to filter.
2) in the old filter screen you saw directly the total of dives corresponding to the criteria. But, in 4.9, I didn't see this total anywhere.
### Operating system:
 Windows 10 Family 1903 64 bits (french )
 

### Subsurface version:
 4.9.3
 No (4.9.1 & 4.8.6)
 official

### Steps to reproduce:

### Current behavior:

### Expected behavior:
 

### Additional information:

### Mentions:

dirkhh, 2019-09-22:
Check the equipment tab.

msd3287, 2019-09-23:
Ok, thanks. 
Didn't see there.

closed 2019-09-23 by msd3287