subsurface/subsurface #2295: Significant performance regression introduced in f8c5c8b

dirkhh, 2019-09-22:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

latest master, at least with Subsurface-mobile, shows a significant performance regression at start with my ~650 dive log

### Operating system:

Tried both with mobile on Desktop under Linux as well as with Android

### Subsurface version:

latest master
a bisect shows that the issue is introduced by f8c5c8bedf

### Steps to reproduce:

start Subsurface=mobile
significant delay (~15-20 seconds) after dive log is opened

### Additional information:

### Mentions:

@bstoeger - this one is yours...

bstoeger, 2019-09-22:
Strange - I get that long wait even with a very short test file. But once it's loaded it seems to behave normally.

dirkhh, 2019-09-22:
Yes, it's just at the start. But the delay on Android is so long that I don't want to have people think that the app is hung at start - that was my first perception when trying. It takes a LONG time where it just sits at the logo.

dirkhh, 2019-09-23:
I'm trying to profile this and am confused.
The gprof output lists a total time of 2.64 seconds for startup - yet it actually takes 28 seconds to start up.
GRMBL
I'm traveling the next 48 hours... I guess I'll have to release a Subsurface-mobile beta with that performance regression...

bstoeger, 2019-09-26:
It happens here:
```
void DiveListModel::reload()
{
        beginResetModel();
        endResetModel();
}
```
For me that generates 60(!) full DiveObjectHelper objects per dive(!).
Of course, the stack dump only shows QML. Brilliant.

bstoeger, 2019-09-26:
Here's the stack back trace:
```
#0  DiveListModel::data (this=0x7fffffffdb80, index=..., role=257) at /home/bs/src/subsurface/qt-models/divelistmodel.cpp:211
#1  0x00007ffff59e4939 in QSortFilterProxyModel::data(QModelIndex const&, int) const () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#2  0x00007ffff65adaad in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5
#3  0x00007ffff66d3044 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5
#4  0x00007ffff66d48c9 in QV4::QObjectWrapper::getQmlProperty(QV4::ExecutionEngine*, QQmlContextData*, QObject*, QV4::String*, QV4::QObjectWrapper::RevisionMode, bool*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5
#5  0x00007ffff66754d3 in QV4::QQmlContextWrapper::get(QV4::Managed const*, QV4::String*, bool*) () from /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5
#6  0x00007ffff663290a in QV4::ExecutionContext::getProperty(QV4::String*) () from /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5
#7  0x00007ffff66eb99d in QV4::Runtime::method_loadName(QV4::ExecutionEngine*, int) () from /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5
#8  0x00007fffd2b3ba56 in ?? ()
#9  0x00007fffe409d410 in ?? ()
#10 0x00007fffe409d420 in ?? ()
#11 0x0000555555a6e8f0 in ?? ()
#12 0x0000000000000000 in ?? ()
```

bstoeger, 2019-09-26:
@dirkhh: Experimentation shows that on model-reset the `mobile-widgets/qml/DiveDetailsView.qml` page is loaded once per dive! That means one profile rendering per dive! I fear I need help here.

bstoeger, 2019-09-26:
I think I got it - apparently for some crazy reason `beginResetModel()`/`endResetModel()` causes the details page to be rendered for *every* dive. You have to remove all rows and then readd all rows. WTF!?

bstoeger, 2019-09-26:
Dirty quickfix in #2303. Please test.

commit eecca6aa (2019-09-27): Mobile: replace model-reset by row-addition in DiveListModel::reload()

Owing to apparent QML breakage, a model-reset leads to the DiveDetail
page being reloaded for every dive in the list(!). Therefore, add
rows instead.

This leads to extremely subtle code, as it is now imperative that
the model has been properly cleared beforehand. Nevertheless, for
now we have to do this to fix a severe performance regression.

Fixes #2295
closed 2019-09-27 by dirkhh