subsurface/subsurface #2301: Subsurface-mobile 2.2.1(4.9.3.30): no menu

hhmeyer, 2019-09-23:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When I start the app, the menu shows itself just for a very brief time (probably less than a second) before the list of dives gets drawn over it.

### Operating system:

Android 8.1

bq Aquaris Pro

### Subsurface version:

Subsurface-mobile 2.2.1(4.9.3.30)

It did work before the update.

official release

### Steps to reproduce:

Open app

### Current behavior:

Menu only show for a very short time

### Expected behavior:

Menu shown constantly

### Additional information:

### Mentions:

dirkhh, 2019-09-23:
This is a new regression on a small number of devices that we are working on fixing.
It appears that as a workaround you can rotate your device sideways, wait for Subsurface-mobile to redraw the screen, and then rotate it back to the original position. So far for everyone encountering this problem that was sufficient to go back to a correctly drawn screen

hhmeyer, 2019-09-23:
I can confirm that this works.

closed 2019-10-28 by dirkhh