subsurface/subsurface #2307: SIGSEGV on exit between v4.8.5 and v4.8.6 (QtNetworkAccessManager)

kristapsdz, 2019-09-29:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Starting v4.8.6, subsurface reliably coredumps on exit (or specifically, atexit(3), har har), regardless of how the exit happens.  The same SIGSEGV occurs in the 4.9.x series.  A full backtrace is included below for v4.8.6.  v4.8.5 does not exhibit the same behaviour and is otherwise rock solid.

### Operating system:

OpenBSD 6.5, amd64.  I can list all libraries used, but they're held constant between the crashing >4.8.5 and non-crashing versions.

### Subsurface version:

All tested versions are downloaded from the main site and use the OpenBSD packaging system, which worked fine for 4.8.5 and before.  This is minimally patched as seen in [ports/misc/subsurface/patches](https://cvsweb.openbsd.org/cgi-bin/cvsweb/ports/misc/subsurface/patches/).

### Steps to reproduce:

Simply start and quit.  I tried this with a new and existing profile with the same results, cloud storage or not.  Starting from the command line.

### Additional information:

The backtrace is as follows:

```
[New process 451359]
[New process 564785]
[New process 480754]
Core was generated by `subsurface'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00000825cc984402 in QExplicitlySharedDataPointer<QNetworkConfigurationPrivate>::~QExplicitlySharedDataPointer() () from /usr/local/lib/libQt5Network.so.2.2
[Current thread is 1 (process 451359)]
(gdb) bt
#0  0x00000825cc984402 in QExplicitlySharedDataPointer<QNetworkConfigurationPrivate>::~QExplicitlySharedDataPointer() () from /usr/local/lib/libQt5Network.so.2.2
#1  0x00000825cc982cdf in QNetworkConfiguration::~QNetworkConfiguration() () from /usr/local/lib/libQt5Network.so.2.2
#2  0x00000825cc8a5cfc in QNetworkAccessManagerPrivate::~QNetworkAccessManagerPrivate() () from /usr/local/lib/libQt5Network.so.2.2
#3  0x00000825cc8a5da6 in QNetworkAccessManagerPrivate::~QNetworkAccessManagerPrivate() () from /usr/local/lib/libQt5Network.so.2.2
#4  0x0000082537f0b69e in QScopedPointerDeleter<QObjectData>::cleanup(QObjectData*) () from /usr/local/lib/libQt5Core.so.2.2
#5  0x0000082537f06fbd in QScopedPointer<QObjectData, QScopedPointerDeleter<QObjectData> >::~QScopedPointer() () from /usr/local/lib/libQt5Core.so.2.2
#6  0x0000082537ef9920 in QObject::~QObject() () from /usr/local/lib/libQt5Core.so.2.2
#7  0x00000825cc8a13e6 in QNetworkAccessManager::~QNetworkAccessManager() () from /usr/local/lib/libQt5Network.so.2.2
#8  0x00000822ea62030e in ImageDownloader::~ImageDownloader (this=0x822ea7ebeb8 <imageDownloader>) at /usr/ports/pobj/subsurface-4.8.6/Subsurface-4.8.6/core/imagedownloader.h:11
#9  0x00000825816149a4 in _libc___cxa_finalize (dso=<optimized out>) at /usr/src/lib/libc/stdlib/atexit.c:177
#10 0x0000082581657a91 in _libc_exit (status=0) at /usr/src/lib/libc/stdlib/exit.c:54
#11 0x00000822ea323149 in ___start ()
#12 0x0000000000000000 in ?? ()
```

If you have suggestions what I can comment out or otherwise try between 4.8.5 and any later version, just let me know and I'll run it.  I've already tried a few things without success in perusing the differences between 4.8.5 and 4.8.6: the "synchronous" block in fetchThumbnail() and a use of SIGNAL in profilewidget2 to no avail.

dirkhh, 2019-09-29:
The stack trace clearly points at the ImageDownloader being the culprit.
I assume you are building from source? In that case `git bisect` might be the easiest way to figure out exactly what change causes this.
Needless to say, I don't see this here (neither on Mac nor Linux)

dirkhh, 2019-10-28:
no response in four weeks, closing for inactivity

closed 2019-10-28 by dirkhh

kristapsdz, 2019-12-14:
I'm sorry for taking so long, but I've finally had the time to track down the exact commit between this versions that causes this issue.  It's commit e685e7e9de3ad93a32fe51d7a04407c574cc08be.

More generally, I found that disabling FBSUPPORT in earlier versions (including stock 4.8.5) exhibits the same behaviour.  Same backtrace as above.  I tracked this down to `MainWindow::setupSocialNetworkMenu` and specifically `FacebookPlugin *facebookPlugin = new FacebookPlugin();`.  With this line (and those depending upon it) commented out, the segfault occurs on exit.  Otherwise, the crash does not occur.

I'm not conversant in C++, but I see that `FacebookPlugin` initialises `FacebookConnectWidget`, which obtains an instance of `FacebookManager`, which initialises a pointer to a `QNetworkAccessManager`.  

Any hints on why this (or rather, not having this) would be causing a segfault on exit?