subsurface/subsurface #2329: Add dive mode to filter function

plxflx, 2019-10-21:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [X] New feature request
- [ ] Discussion request

### Issue long description:
Can we please add a "Dive mode" field to the dive filter function?

### Operating system:
Windows and Ubuntu

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2019-10-28:
I haven't seen a lot of requests for this. This would not be excessively hard to do, but it will clutter up the filter screen a bit. And I think the vast majority of users will never use this as most people only ever dive OC.
I'm not going to close this just yet. If we get more requests on this issue, I'll consider it. But right now I'm leaning towards 'no'.

plxflx, 2019-10-30:
I think its doable. I guess the majority of the user base are OC divers but as there is the options for CCR and free dives (and pSCR) I think this should be supported all the way. Further CCR training often requires a certain number of hours on the unit. The ability to filter this out make it possible to show this to an instructor. The same goes for continued cave/mine diving training.

A simple layout would be to just add the same drop down from log header to filter layout and just add the option of 'all'.

adamel, 2019-11-03:
This would be very useful.

bstoeger, 2020-05-03:
Should be in release (even though UI is not perfect). Closing.

closed 2020-05-03 by bstoeger