subsurface/subsurface #2331: Request to calculate SAC separately for each gas

Fscked-In-The-Head, 2019-10-22:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x ] New feature request
- [ ] Discussion request

### Issue long description:

Request to calculate SAC per gas, rather than for the entire dive.

When planning a dive, you can specify a different deco SAC than working SAC. But without an AI computer there's no way to accurately determine those within Subsurface. This would be possible to do for backgas deco by entering SPG readings taken during the dive, but a simpler way could be calculating SAC per gas. For example, if the working phase of a dive was on backgas, and deco was on a different mix, calculating SAC per gas would effectively produce working and deco SAC values.

Ideally this deco SAC measurement would also appear in the statistics to allow viewing trends.

This addition should have no effect on recreational single-gas dives, but would allow for more accurate logging and planning for technical multi-gas dives.

glance-, 2019-10-23:
Per gas SAC is already calculated, and it's in the information tab, the SAC box.

Subsurface doesn't know whats an "deco" gas or an "bottom" gas, so its hard to provide statistics based on those metrics.

Fscked-In-The-Head, 2019-10-23:
My mistake, I wasn't aware that was already done. I was looking at the SAC posted in the dive list.

Thank you!

closed 2019-10-23 by Fscked-In-The-Head