subsurface/subsurface #2350: Subsurface-mobile: after import from BLE dive computer the app is not responding

snapiri, 2019-10-28:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Importing dives from dive computer via BLE, the import succeeds, but then when the dive list should be redrawn, the app gets unresponsive. The only solution was to kill it, and then reopen it. The new dives appear in the list.

### Operating system:

OS: Android 8.0.0

Device: Xiaomi Mi5

### Subsurface version:

Subsurface-mobile:2.2.3(4.9.3.192)

AFAIK this is a new issue

This is a beta version of the mobile app.

### Steps to reproduce:

1) Connect dive computer via bluetooth
2) Download new dives
3) Import the dives
4) App becomes unresponsive

### Current behavior:

App is unresponsive

### Expected behavior:

Dives list should be redrawn and functionality should resume.

### Additional information:

log file attached (Note line 312)
[subsurface.log](https://github.com/Subsurface-divelog/subsurface/files/3780082/subsurface.log)

### Mentions:

dirkhh, 2019-10-28:
Those odd QPainter errors have started showing up with the new version of Kirigami that we switched to a few weeks ago - I haven't been able to tell if there are any visible side effects, though.

The current beta is actually 4.9.3.211, but the behavior that you are describing should already be fixed in the previous one that you appear to be running.

The good news is that according to the log the dives were actually saved to local storage before things got bad, so at least you don't have to re-download. If this still happens with the latest beta please let me know - I have done about a dozen Android downloads with the Shearwater Teric (quite similar to the Perdix from a BLE perspective) on Android this week and thought that this bug was fixed...

snapiri, 2019-10-29:
Unfortunately this also happens also in the latest beta version.If more logs or tests are required please let me know.

dirkhh, 2019-10-29:
That's... disappointing. Especially since I can't reproduce it. And the log doesn't give me a lot to go on.
All I can think of right now is to possibly add a bunch more debugging output to a test version to get more information how far we get before this happens.
Can you side load APKs on your device? I.e., if I point you to a signed APK on a website, can you download and install that? This way I wouldn't have to flood all the beta testers with versions specific to you...

snapiri, 2019-10-29:
No problem. Can do that...

dirkhh, 2019-10-29:
Can you please give this one a try? Download dives from your dive computer. Attach  subsurface.log (even if it succeeds) to this issue. Thanks

https://subsurface-divelog.org/downloads/test/Subsurface-mobile-4.9.3.242-arm64.apk

snapiri, 2019-10-30:
The import succeeded and the app did not get stuck...Log attached.

https://subsurface-divelog.org/downloads/test/Subsurface-mobile-4.9.3.242-arm64.apk

snapiri, 2019-10-30:
After trying the new version I think there is something different I did with it.
With playstore version, the dive did not show on the list after the import, so I refreshed the list (by pulling the list down), this is when it got stuck.
With the new version, the imported dive appeared, so there was no need to refresh.
I tried again, this time **with** refreshing and it still succeeded.
Another difference is that now it pauses for a short while on the spinner while adding the dives.

dirkhh, 2019-10-30:
Yes, this version does things slightly differently in order to reduce the risk of the UI getting stuck. Your information seems to show that this worked! Thank you so much for your response.
I'll clean this code up a little and push it into our sources - in a day or two this should hit our beta version. But I hope you're ok if I close this issue for now.

closed 2019-10-30 by dirkhh

snapiri, 2019-10-30:
Thanks for your quick update :+1: