subsurface/subsurface #2352: Zero values in Dive planner

tormento, 2019-10-29:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Subsurface dive planner creates deco stops with zero minutes value. It is twice a bug:

1) it creates deeper deco stops, i.e. 21, omits the one at 18 and restart at 15 (usually the deco stops are contiguous unless some ascent speed has been changed during ascent)
2) there should be no 0 minutes deco stops: perhaps some rounding error

Runtime: 63min

  | depth | duration | runtime | gas
-- | -- | -- | -- | --
➘ | 50m | 3min | 3min | air
➙ | 50m | 25min | 27min |  
➚ | 21m | 3min | 30min |  
- | 21m | 1min | 31min | EAN50
**➚ | 18m | 0min | 32min |  
- | 18m | 0min | 32min |**  
➚ | 15m | 0min | 32min |  
- | 15m | 2min | 34min |  
➚ | 12m | 0min | 34min |  
- | 12m | 3min | 37min |  
➚ | 9m | 0min | 37min |  
- | 9m | 4min | 41min |  
➚ | 6m | 0min | 41min |  
- | 6m | 21min | 62min |  
➚ | 0m | 1min | 63min |  

### Operating system:

Windows 10 x64

### Subsurface version:

4.9.3.235

### Steps to reproduce:

27' @ 50m salt water
D12 air + AL50 EAN50
ascent 9 / descent 20
GF 45/95
v drop to first depth
v last stop at 6m
v only switch at required stops
bottom SAC 20 deco 18

### Current behavior:

Runtime: 63min

  | depth | duration | runtime | gas
-- | -- | -- | -- | --
➘ | 50m | 3min | 3min | air
➙ | 50m | 25min | 27min |  
➚ | 21m | 3min | 30min |  
- | 21m | 1min | 31min | EAN50
**➚ | 18m | 0min | 32min |  
- | 18m | 0min | 32min |**  
➚ | 15m | 0min | 32min |  
- | 15m | 2min | 34min |  
➚ | 12m | 0min | 34min |  
- | 12m | 3min | 37min |  
➚ | 9m | 0min | 37min |  
- | 9m | 4min | 41min |  
➚ | 6m | 0min | 41min |  
- | 6m | 21min | 62min |  
➚ | 0m | 1min | 63min |  

### Expected behavior:

Correct value at 18 or modified deco stops at 21 to zero too and moving to 18 that one.

### Additional information:

### Mentions:

atdotde, 2019-10-29:
I fixed this already in 00c902d66ca114836846132ba775e51e8003ab56, please see #2282.

tormento, 2019-10-29:
Sorry but **NOT FIXED**

As you can see, with latest continuous, the issue is still present.

tormento, 2019-10-29:
Round up 0-29 seconds (ascending 3 meters is 20 seconds at 9 m/s) to zero and sum the time amount to deco stop (20+26=46) then round the range 30-59 to 1 full minute in the deco stop.

bstoeger, 2020-05-03:
Is this fixed?

atdotde, 2020-05-03:
I guess so

closed 2020-05-03 by atdotde