subsurface/subsurface #2358: Subsurface Mobile "Show all bluetooth devices..." is not sticky over reboots

giffenk, 2019-11-03:
### Describe the issue:
- [ x] Bug
After a reboot, the bluetooth setting "Show all devices..." is unchecked.

### Issue long description:

In order to get a list of bluetooth devices to show up in the download screen you may need to enable this option. All known bluetooth devi ec then appear in the connection list. After a reboot of the phone, this option is turned off and no bluetooth devices appear in the connection list.

Note that in my world none of the bluetooth devices were "recognized" dive computers. this is a secondary issue. the real problem is that the option turned itself off, leading to inconsistent behaviour for the end user.
 
### Operating system:

Android 10

Essential PH-1

### Subsurface version:

Subsurface mobile 2.2.2 (4.9.3.119)

Subsurface mobile downloaded from android store thingy

### Steps to reproduce:

set the option
restart phone
option is no longer set...
### Current behavior:

### Expected behavior:

Option should be retained over restarts
### Additional information:

### Mentions:

dirkhh, 2019-11-03:
That is by design. Showing all devices shouldn't be the user experience as that is extremely hostile with the meaningless UUIDs shown. The goal is to make sure we recognize all dive computers and show only the dive computers.
If we don't recognize your dive computer, please start a session on Subsurface-mobile, download from your dive computer (even if it says no new dives found), then go to the About page, copy the log, and paste it into a response to this ticket so we can fix the recognition issue.

giffenk, 2019-11-05:
I understand the reasoning. BUT: this behaviour is confusing as this appears to be the only "non sticky" setting. So it is inconsistent with all other settings. And hence it causes confusion. This leads the user to start to worry about what other settings that I made will get undone without telling me? Inconsistent behaviour causes users to lose confidence in the quality of the software.
 
If the current behaviour is retained maybe it is worthwhile to alter the description of this setting to BOLDLY indicate to the user that this option is transient (i.e. it is very special and works differently from everything else) and will automatically be turned off when ... (I am not sure when...).

Cheers...

dirkhh, 2019-11-06:
That's a valid point. I'll see if I can fit this in the settings screen as a clarification.

dirkhh, 2019-11-21:
BTW: which dive computer do you use that requires this setting?

closed 2019-11-23 by dirkhh

giffenk, 2020-02-11:
Sorry - I did not notice your question asking about the dive computer. The computers are various models of Uwatec Aladins. 
Since subsurface mobile prevents them from connecting over bluetooth I abandoned use of Subsurface mobile. They appear to be fully supported by Subsurface desktop on Windows since Bluetooth connections are treated like serial connections.

commit 838548a8 (2019-11-21): Mobile: make it clear that 'all BT devices' setting is temporary

Making this persistant creates a very unpleasant user experience.
And if a dive computer needs this setting, we really want people
to report this to us so we can fix it.

Fixes: #2358
commit d86f7b56 (2019-11-21): Mobile: make it clear that 'all BT devices' setting is temporary

Making this persistant creates a very unpleasant user experience.
And if a dive computer needs this setting, we really want people
to report this to us so we can fix it.

Fixes: #2358
commit e2572c9e (2019-11-22): Mobile: make it clear that 'all BT devices' setting is temporary

Making this persistant creates a very unpleasant user experience.
And if a dive computer needs this setting, we really want people
to report this to us so we can fix it.

Fixes: #2358
PR #2392 (2019-11-23): Bt updates

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

two small fixes.
One subtle update to the detection code for the OSTC family of computers - picking OSTC 3 ended up causing us to not recognize certain BLE versions of the OSTC on iOS as dive computers.
And a change to the language on the settings page that clarifies that the "show all BT devices" settings is not persistent.

### Related issues:

Fixes #2358 

# [... cut]