subsurface/subsurface #2362: QT error build using android build wrapper

StephenGoodall, 2019-11-08:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [x] Discussion request

### Issue long description:

When running the android build wrapper script a warning about credentials comes up from the Qt install step I think, is it fine to ignore this or do i need to get some credentials in order to build it?

Qt installation found, backing it up to Qt_OLD.
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-******'
[3711] Warning: No QtAccount credentials found. Please login via the maintenance tool of the SDK.

### Operating system:

WSL - Ubuntu - just to build the APK

StephenGoodall, 2019-11-08:
I guess this syntax error is kind of related too, but might be an issue from libxml2 rather than any of the subsurface bits:

Checking zlib
../libxml2/configure: line 13070: syntax error near unexpected token `Z,zlib,'
../libxml2/configure: line 13070: `    PKG_CHECK_MODULES(Z,zlib,'
ls: cannot access './subsurface-mobile-build-arm/build/outputs/apk/debug/*.apk': No such file or directory

Has anyone come up against this one and resolved it? 😄

dirkhh, 2019-11-08:
you can ignore the credentials warning, same for the XDG_RUNTIME warning.

the Android build is tested with every single push to GitHub - so I'd be surprised if there was something fundamentally broken. And of course I use it here locally every time I build new binaries.

I don't recall ever seeing the `configure` error in libxml2 that you describe.

StephenGoodall, 2019-11-08:
thanks @dirkhh that one seemed to be because i didn't have pkg-config installed before i ran the wrapper script the first time

I'm coming up against something with the cMake bits I think (I don't fully understand how that all links together)
Does the automated build use the same versions (NDK is r18b, Qt is 5.13.1)? I'm assuming they do as the script grabbed those versions for me 😄 

The error is:
`In file included from /home/stephen/subsurface/profile-widget/profilewidget2.cpp:36:
In file included from //home/stephen/Qt/5.13.1/android_armv7/include/QtCore/qmath.h:55:
/home/stephen/android-ndk-r18b/sources/cxx-stl/llvm-libc++/include/cmath:321:9: error: no member named 'isless' in the global namespace
using ::isless;
      ~~^
/home/stephen/android-ndk-r18b/sources/cxx-stl/llvm-libc++/include/cmath:322:9: error: no member named 'islessequal' in the global namespace
using ::islessequal;
      ~~^
/home/stephen/android-ndk-r18b/sources/cxx-stl/llvm-libc++/include/cmath:323:9: error: no member named 'islessgreater' in the global namespace
using ::islessgreater;
      ~~^
/home/stephen/android-ndk-r18b/sources/cxx-stl/llvm-libc++/include/cmath:324:9: error: no member named 'isunordered' in the global namespace
using ::isunordered;
      ~~^
/home/stephen/android-ndk-r18b/sources/cxx-stl/llvm-libc++/include/cmath:325:9: error: no member named 'isunordered' in the global namespace
using ::isunordered;
      ~~^
11 errors generated.
make[2]: *** [profile-widget/CMakeFiles/subsurface_profile.dir/profilewidget2.cpp.o] Error 1
make[1]: *** [profile-widget/CMakeFiles/subsurface_profile.dir/all] Error 2
make: *** [all] Error 2`

So it seems to not be able to see some of the bits from the math classes, reading around, it seems to be something like this:
https://github.com/termux/termux-packages/issues/1149

Maybe I just need to wipe the folder out and run it again from scratch......

glance-, 2019-11-08:
I saw that one, and i "fixed" it locally by commenting out the "problematic" lines =)

StephenGoodall, 2019-11-08:
Glad it wasn't just me, I'll have another go.
When you say problematic lines, I'm assuming that means the "include" lines (are these roughly equivalent to imports in java?) rather than the lines that are trying to use "isless" etc?

Wonder if it could be down to a difference in the cmake version that is on my Ubuntu - 3.10.2 I think, compared to the build server. It's more than likely something to do with my setup....it usually is!

StephenGoodall, 2019-11-08:
Commented out the relevant "using" lines and it's built the APK now - thanks for the tips both of you 😁 

And thanks for the build wrapper - simplifies things a lot for people like me 👍

closed 2019-11-08 by StephenGoodall