subsurface/subsurface #2365: Pro Plus X not connecting or downloading

petergraemerivers, 2019-11-10:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Pro Plus X not connecting or downloading by USB or BLE to Subsurface Mobile
### Operating system:

Android v9 4.4.153

Samsung Tab S4

Yes. Windows full version. Similar issues, no dives download
### Subsurface version:

2.2.3(4.9.3.279)

Has intermittantly worked in the past on previous versions. Not sure which though sorry.

Official release from playstore - but says it's Beta?

### Steps to reproduce:

1) pair pro plus x with Tab S4
2) select download from DC
3) select vendor, computer, and LE connection and click download
### Current behavior:

It says "Info:Connecting ..." and seems to lock up. No dives are downloaded.

DC downloads using manufacturers software - Diverlog
### Expected behavior:

Dives download
### Additional information:

Log File

---------- subsurface.log ----------
"0.012: Failed to open logfile /storage/emulated/0/subsurface.log at Sun Nov 10 16:45:47 2019 error: Permission denied"
"0.014: Failed to open logfile /storage/emulated/0/Documents/subsurface.log at Sun Nov 10 16:45:47 2019 error: Permission denied"
"0.014: Successfully opened logfile /storage/emulated/0/Android/data/org.subsurfacedivelog.mobile/files/Documents/subsurface.log at Sun Nov 10 16:45:47 2019"
"0.015: Starting Subsurface-mobile:2.2.3(4.9.3.273):Android  (9.0):arm64:en-NZ"
"0.015: built with libdivecomputer v0.7.0-devel-Subsurface-NG (426a39fc7378cac3f5b1d3983be38d56ab526e78)"
"0.015: built with Qt Version 5.13.1, runtime from Qt Version 5.13.1"
"0.015: built with libgit2 0.26.0"
"0.015: Running on Android  (9.0)"
"0.015: SM-T830/samsung/gts4lwifixx"
BTDiscoveryReDiscover: localBtDevice.isValid() true
"BTDiscoveryReDiscover: localDevice Galaxy Tab S4 is powered on, starting discovery"
paired BLE device type 2 with address "LE:E0:00:00:00:06:43"
Found new device: "Pro Plus X" "LE:E0:00:00:00:06:43"
"this could be a Oceanic Pro Plus X"
Paired = "ER001428" "LE:E0:00:00:00:06:43"
"0.039: Created position source android"
"0.039: Set GPS service update interval to 300 s"
"0.039: location service is not available"
qrc:/org/kde/kirigami/Page.qml:371:9: QML Loader: Binding loop detected for property "active"
"0.090: Window width changed to 780 orientation 2"
"0.090: Screen rotated, no action necessary"
"0.090: 1 columns with column width of 780"
"0.090: width in Grid Units 30 original gridUnit 26 now 26"
"0.090: Done setting up sizes"
"0.423: Switching to no cloud mode"
"0.440: working in no-cloud mode"
"0.455: Unable to look up revision 'master'"
[... 91 lines cut ...]
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::setParent: Cannot set parent, new parent is in a different thread
"550.505: AppState changed to active with no save ongoing and no unsaved changes"
qrc:/org/kde/kirigami/templates/private/ScrollView.qml:103:9: QML Flickable: Binding loop detected for property "contentHeight"
"553.423: pageStack switched to "
qrc:/org/kde/kirigami/templates/private/ScrollView.qml:103:9: QML Flickable: Binding loop detected for property "contentHeight"
"562.327: AppState changed to inactive with no save ongoing and no unsaved changes"
"562.355: AppState changed to suspended with no save ongoing and no unsaved changes"
"594.442: AppState changed to active with no save ongoing and no unsaved changes"
"598.163: AppState changed to inactive with no save ongoing and no unsaved changes"
"598.188: AppState changed to suspended with no save ongoing and no unsaved changes"
"825.432: AppState changed to active with no save ongoing and no unsaved changes"
---------- finish ----------

### Mentions:

dirkhh, 2019-11-10:
If the play store tells you 'beta' that means that you joined our 'open beta' program. You should be able to simply leave that from within the Google Play app - towards the bottom of the Subsurface-mobile page. We (the developers) have no control over how Google handles that.

Your Android device thinks that it found the ProPlus X - but some of the BLE devices themselves remember which phone / computer they are paired with so they might refuse to talk to the Android phone. One thing that sometimes helps is to forget the device in Android and pair again.

petergraemerivers, 2019-11-10:
Thanks Dirk. I have now removed myself from the beta programme and reinstalled subsurface mobile. Also re-paired computer. I still have the same problem. I wondered if the fact that i have more dives in subsurface than on the computer may be messing things up.  Thanks for your help,  i love the program and three DC, but getting between them has been problematic.

New log
---------- subsurface.log ----------
"0.029: Failed to open logfile /storage/emulated/0/subsurface.log at Mon Nov 11 07:16:13 2019 error: Permission denied"
"0.030: Failed to open logfile /storage/emulated/0/Documents/subsurface.log at Mon Nov 11 07:16:13 2019 error: Permission denied"
"0.030: Successfully opened logfile /storage/emulated/0/Android/data/org.subsurfacedivelog.mobile/files/Documents/subsurface.log at Mon Nov 11 07:16:13 2019"
"0.031: Starting Subsurface-mobile:2.2.2(4.9.3.119):Android  (9.0):arm64:en-NZ"
"0.031: built with libdivecomputer v0.7.0-devel-Subsurface-NG (ce6d9896a79afaa82641132e338f8744714c8593)"
"0.031: built with Qt Version 5.13.1, runtime from Qt Version 5.13.1"
"0.031: built with libgit2 0.26.0"
"0.031: Running on Android  (9.0)"
"0.031: SM-T830/samsung/gts4lwifixx"
BTDiscoveryReDiscover: localBtDevice.isValid() true
"BTDiscoveryReDiscover: localDevice Galaxy Tab S4 is powered on, starting discovery"
paired BLE device type 2 with address "LE:E0:00:00:00:06:43"
Found new device: "Pro Plus X" "LE:E0:00:00:00:06:43"
"this could be a Oceanic Pro Plus X"
Paired = "ER001428" "LE:E0:00:00:00:06:43"
"0.071: Created position source android"
"0.071: Set GPS service update interval to 300 s"
"0.071: location service is not available"
qrc:/org/kde/kirigami/Page.qml:276:5: QML Loader: Binding loop detected for property "active"
Window width changed to 780 orientation 2
Screen rotated, no action necessary
Model size of -1 is less than 0
"0.639: no cloud credentials"
Window width changed to 1280 orientation 2
first real change, so recalculating units and recording size as 1280 x 728
Column width 1280 root item width 1280
width in Grid Units 49 original gridUnit 26 now 26
Done setting up sizes
setting the defaultColumnWidth to 546
checkPendingIntents 
SubsufaceTheme constructor completed, initial width 1280
Already adjusted size, ignoring this
Using the following font: Roboto at 21pt with mobile_scale: 1
loaded main.qml
qqwindow devicePixelRatio 2 2
qml_window reports width as 1280 associated screen width 1280 Qt screen reports width as 1280
Supported dive computers:
"Aeris: 500 AI (SERIAL), A300 (SERIAL), A300 AI (SERIAL), A300CS (SERIAL), Atmos 2 (SERIAL), Atmos AI (SERIAL), Atmos AI 2 (SERIAL), Compumask (SERIAL), Elite (SERIAL), Elite T3 (SERIAL), Epic (SERIAL), F10 (SERIAL), F11 (SERIAL), Manta (SERIAL), XR-1 NX (SERIAL), XR-2 (SERIAL)"
"Aqualung: i200 (SERIAL), i200C (SERIAL, BLE), i300 (SERIAL), i300C (SERIAL, BLE), i450T (SERIAL), i550 (SERIAL), i550C (SERIAL, BLE), i750TC (SERIAL, BT), i770R (SERIAL, BLE)"
"Atomic Aquatics: Cobalt (USB), Cobalt 2 (USB)"
"Beuchat: Mundial 2 (SERIAL), Mundial 3 (SERIAL), Voyager 2G (SERIAL)"
"Cochran: Commander I (SERIAL), Commander II (SERIAL), Commander TM (SERIAL), EMC-14 (SERIAL), EMC-16 (SERIAL), EMC-20H (SERIAL)"
"Cressi: Cartesio (SERIAL), Drake (SERIAL), Giotto (SERIAL), Goa (SERIAL), Leonardo (SERIAL), Newton (SERIAL)"
"Deepblu: Cosmiq+ (BLE)"
"DiveSystem: Orca (SERIAL), iDive DAN (SERIAL), iDive Deep (SERIAL), iDive Easy (SERIAL), iDive Free (SERIAL), iDive Pro (SERIAL), iDive Reb (SERIAL), iDive Stealth (SERIAL), iDive Tech (SERIAL), iDive X3M (SERIAL)"
"Garmin: Descent Mk1 (USBSTORAGE)"
"Genesis: React Pro (SERIAL), React Pro White (SERIAL)"
"Heinrichs Weikamp: Frog (SERIAL, BT), OSTC (SERIAL), OSTC 2 (SERIAL, BT, BLE), OSTC 2 TR (SERIAL, BT, BLE), OSTC 2C (SERIAL), OSTC 2N (SERIAL), OSTC 3 (SERIAL), OSTC 4 (SERIAL, BT, BLE), OSTC Mk2 (SERIAL), OSTC Plus (SERIAL, BT, BLE), OSTC Sport (SERIAL, BT, BLE), OSTC cR (SERIAL)"
"Hollis: DG02 (SERIAL), DG03 (SERIAL), TX1 (SERIAL)"
"Mares: Genius (SERIAL, BLE), Puck Pro (SERIAL, BLE), Quad (SERIAL, BLE), Quad Air (SERIAL, BLE), Smart (SERIAL, BLE), Smart Air (SERIAL, BLE), Smart Apnea (SERIAL, BLE)"
"Oceanic: Atom 1.0 (SERIAL), Atom 2.0 (SERIAL), Atom 3.0 (SERIAL), Atom 3.1 (SERIAL), Datamask (SERIAL), F10 (SERIAL), F11 (SERIAL), Geo (SERIAL), Geo 2.0 (SERIAL), Geo 4.0 (SERIAL, BLE), OC1 (SERIAL), OCS (SERIAL), OCi (SERIAL), Pro Plus 2 (SERIAL), Pro Plus 2.1 (SERIAL), Pro Plus 3 (SERIAL), Pro Plus X (SERIAL, BLE), VT 4.1 (SERIAL), VT Pro (SERIAL), VT3 (SERIAL), VT4 (SERIAL), VTX (SERIAL), Veo 1.0 (SERIAL), Veo 180 (SERIAL), Veo 2.0 (SERIAL), Veo 200 (SERIAL), Veo 250 (SERIAL), Veo 3.0 (SERIAL), Versa Pro (SERIAL)"
"Ratio: iDive Color Deep (SERIAL), iDive Color Easy (SERIAL), iDive Color Fancy (SERIAL), iDive Color Free (SERIAL), iDive Color Pro (SERIAL), iDive Color Reb (SERIAL), iDive Color Tech+ (SERIAL), iDive Deep (SERIAL), iDive Easy (SERIAL), iDive Fancy (SERIAL), iDive Free (SERIAL), iDive Pro (SERIAL), iDive Reb (SERIAL), iDive Tech+ (SERIAL), iX3M GPS Deep (SERIAL, BT), iX3M GPS Easy (SERIAL, BT), iX3M GPS Fancy (SERIAL, BT), iX3M GPS Pro  (SERIAL, BT), iX3M GPS Reb (SERIAL, BT), iX3M GPS Tech+ (SERIAL, BT), iX3M Pro Deep (SERIAL), iX3M Pro Easy (SERIAL), iX3M Pro Fancy (SERIAL), iX3M Pro Pro (SERIAL), iX3M Pro Reb (SERIAL), iX3M Pro Tech+ (SERIAL)"
"Scubapro: Aladin H Matrix (BLE), Aladin Sport Matrix (BLE), Aladin Square (USBHID), G2 (USBHID, BLE), G2 Console (USBHID, BLE), G2 HUD (USBHID, BLE)"
[... cut ...]
INFO: Write: size=14, data=CD40010AE500000104020800000F
INFO: Read: size=5, data=CDC00101A5
ERROR: Result too small: got 1 bytes, expected at least 1 bytes [in /android/subsurface/libdivecomputer/src/oceanic_atom2.c:819 (oceanic_atom2_ble_transfer)]
INFO: Write: size=14, data=CD40010AE500000104020800000F
INFO: Read: size=5, data=CDC00101A5
ERROR: Result too small: got 1 bytes, expected at least 1 bytes [in /android/subsurface/libdivecomputer/src/oceanic_atom2.c:819 (oceanic_atom2_ble_transfer)]
---------- finish ----------

dirkhh, 2019-11-11:
No, the log seems to indicate that there's a communication issue. OK, the error message is a bit weird :-)
@jefdriesen - any idea what's going on here?

jefdriesen, 2019-11-12:
We get a NAK byte (A5) instead of an ACK byte (5A) in response to the authentication command (E5). Maybe this means the Pro Plus X doesn't need the authentication step and doesn't understand the command?

petergraemerivers, 2019-11-19:
Hi gents, wondering if there is any way forward with this issue? 
Regards

jefdriesen, 2020-02-28:
Fixed in commit 1d235daf307f160ede20d2962e9a799954df00f4

petergraemerivers, 2020-02-29:
Wonderful thanks.  Hoping for a dive tomorrow so will check.

mikeller, 2026-10-17:
Hi @petergraemerivers,

Good news: jefdriesen (libdivecomputer maintainer) traced the exact cause from your log, your Pro Plus X replies with a NAK instead of the expected ACK to the authentication command, because it doesn't need/understand that step at all. Fixed in libdivecomputer commit 1d235daf307f160ede20d2962e9a799954df00f4.

Closing this as fixed. Please reopen if your Pro Plus X still hangs on "Connecting..." on a current build.

closed 2026-10-17 by mikeller