subsurface/subsurface #2370: Bluetooth doesn't work in the appimage

glance-, 2019-11-11:
```
$ ./Subsurface.AppImage 
Cannot find a running Bluez. Please check the Bluez installation.
```
### Describe the issue:
- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When running subsurface from the app image, bluetooth doesn't work.

### Operating system:
Tested on Ubuntu 18.04

### Subsurface version:
Tested with v4.9.3-304-g80f123ce2aac

### Steps to reproduce:
Start app image, Import from dive computer, Choose Bluetooth download mode.

### Current behavior:
Bluetooth doesn't work. "Not available"
Says `Cannot find a running Bluez. Please check the Bluez installation.` on commandline.

### Expected behavior:
I expect to see devices on the Discovered devices dialog.

### Additional information:
We get the dbus library from the applimage, and the compiled in default for the dbus socket points to `unix:path=/usr/local/var/run/dbus/system_bus_socket` . I don't know of a single system where that will be right. A better default would be /var/run/dbus/system_bus_socket.

Even when setting the env-var DBUS_SYSTEM_BUS_ADDRESS=/var/run/dbus/system_bus_socket it behaves the same way, but with strace I see some talk on dbus, but nothing about bluez. What else do Qt5Bluetooth need to talk to bluez?

### Mentions:
@probonopd

probonopd, 2019-11-11:
Hmm, I have never seen this error before, maybe https://forum.qt.io/topic/74833/debugging-bluetooth/6 can provide some insight.

glance-, 2019-11-11:
I haven't managed to get Qt to log anything at all about bluetooth, so there's something fishy here...

probonopd, 2019-11-11:
Does Bluetooth need special Qt plugins to work? Then those need to be deployed into the AppImage, too.

glance-, 2019-11-12:
Ok, the libdbus-1.so.3 library is b0rk. Where do that come from?

I just tried to unpack the app-image and replaced the libdbus-1.so.3 lib with the one from trusty and then everything started to work.

dirkhh, 2019-11-12:
That's promising.
The lib should be from the host container the AppImage is built on. Replacing it should be relatively straight forward. Key is to find one that doesn't cause other dependency issues...
What happens if you just delete it in the AppImage? Is the one from your host OS picked?

glance-, 2019-11-12:
Found it:
https://github.com/Subsurface-divelog/subsurface/blob/master/scripts/docker/trusty-qt512/Dockerfile#L53

Why do we need a new dbus? The one in trusty works just fine.

dirkhh, 2019-11-21:
> Found it:
> https://github.com/Subsurface-divelog/subsurface/blob/master/scripts/docker/trusty-qt512/Dockerfile#L53
> 
> Why do we need a new dbus? The one in trusty works just fine.

Sorry for the slow response - working on too many things at once.
My only possible answer here is "because when I worked on it I ran into some issue that I no longer remember and the fix for that issue was to build a new dbus library?"
I honestly don't recall.
To make sure I understand your point - have you tried building the container without that new libdbus and were able to create a working AppImage that way? Or have you just removed the libdbus from the AppImage and that's how you got to the conclusion that the one from Trusty would indeed work?
As I am typing this, I'm thinking that my issue was that I couldn't compile Qt on Trusty with the libdbus that was shipping with it...

glance-, 2019-11-21:
I've just tested to yank out the dbus lib from the app-image so the one from my bionic host os is used, and I've also tested to replace it with the one from trusty, and both works.

We're not even building Qt, we're just using the pre-built official binaries...

glance-, 2019-11-21:
If it weren't for the fact that docker hates my github username (docker is of the opinion that - is a evil character) I could have tested the whole thing automatically....

Subsurface-divelog, 2019-11-21:
> I've just tested to yank out the dbus lib from the app-image so the one from my bionic host os is used, and I've also tested to replace it with the one from trusty, and both works.

It should be fairly simple to remove the dbus library from the AppImage. I can look into that.

> We're not even building Qt, we're just using the pre-built official binaries...

I'm clearly getting old. Or not getting enough sleep. Yes - initially I was trying to build our own Qt, but then I realized that the Qt-distribution bits still worked on Trusty, and so I ended up just building QtWebKit. The order of things in the Dockerfile makes me wonder about all of this, anyway - but this is such a painfully slow process that it tends to turn into a major time-suck whenever I try to make changes there...

commit 05dea82f (2019-11-21): AppImage: don't bundle our libdbus

This appears to prevent Bluetooth from working.

Fixes: #2370

Suggested-by: Anton Lundin <glance@acc.umu.se>
commit 84eb4e48 (2019-11-21): AppImage: don't bundle our libdbus

This appears to prevent Bluetooth from working.

Fixes: #2370

Suggested-by: Anton Lundin <glance@acc.umu.se>
PR #2394 (2019-12-07): AppImage: don't bundle our libdbus
### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

simply delete the offending library before bundling

### Related issues:

Fixes: #2370

### Additional information:

### Release note:

DONE

### Mentions:

@glance- can you please check that the resulting AppImage works for you?
Since this is a PR, you'll need to go to the Action for Linux Qt5.12 and download it from th [... cut]
commit 84394adb (2019-12-07): AppImage: don't bundle our libdbus

This appears to prevent Bluetooth from working.

Fixes: #2370

Suggested-by: Anton Lundin <glance@acc.umu.se>
closed 2019-12-07 by dirkhh