subsurface/subsurface #2376: Dive information tab shows meaningless and wrong values for SAC and gas consumption

sfuchs79, 2019-11-12:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

By clicking around in my dive list I can randomly produce insane values for SAC and gas consumption in the dive information tab.

Examples:
[image]
[image]
[image]

These are all screenshots from the very same dive. Produced by unselecting/selecting this dive. The wrong values happen for gases, where there is no cylinder pressure info at all. So expected would be having printed no info at all.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@dirkhh @bstoeger @willemferguson

willemferguson, 2019-11-13:
Stefan, thank you, this is valuable. Please email me directly one of your dives and the resulting information tab?

willemferguson, 2019-11-13:
@bstoeger, would it also make sense to use  dive->cylinders.nr instead of MAX_CYLINDERS as you did to solve the cylinder pressure pressure graph problem?

bstoeger, 2019-11-13:
All MAX_CYLINDERS instances were removed. Fix is hopefully in #2378.

To test, you just need a dive without begin and/or end-pressure (and no events from an AI computer).

PR #2378 (2019-11-16): Core: set gas use to zero if no values are given
In get_gas_used() the use was left uninitialized if there are neither
user- nor computer-supplied values. This gives random SACs in the UI.
Initialize to 0.

Fixes #2376.

Reported-by: Stefan Fuchs <sfuchs@gmx.de>

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

This is supposed to fix #2376. All info in the commit message.
### Mentions:

@sfuchs79 @willemferguson
commit 065559c3 (2019-11-16): Core: set gas use to zero if no values are given

In get_gas_used() the use was left uninitialized if there are neither
user- nor computer-supplied values. This gives random SACs in the UI.
Initialize to 0.

Fixes #2376.

Reported-by: Stefan Fuchs <sfuchs@gmx.de>
closed 2019-11-16 by bstoeger