subsurface/subsurface #2377: Color of pressure graph

sfuchs79, 2019-11-12:
### Describe the issue:

- [x] Question - maybe a bug ;-)
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

This is for the first moment more a question:
What decides about the color of the pressure graph in the profile? Is it s.th. related to SAC?

I recognized, that for many dives my pressure line is all red, example:
[image]

And I thought this was not the case with older versions of Subsurface.

I have to add that I don't use a air integrated dc and never used one. So pressure info in my dives is always derived from manually entered values.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

bstoeger, 2019-11-12:
Certainly a fallout of the MAX_CYLINDERS removal.

jb2cool, 2020-03-08:
Do your cylinders have a size associated with them? Without an internal volume Subsurface can't work out consumption (the colour is based on consumption rate.

[image]
[image]

mikeller, 2026-10-17:
Hi @sfuchs79,

To close the loop on this: the profile's pressure-line colour reflects computed gas-consumption rate (SAC), which needs a cylinder with a defined size/volume to calculate at all, without one set, Subsurface can't compute consumption and defaults to showing red. This isn't a bug so much as a side effect of a prior cylinder-defaults refactor changing how often dives end up with an undefined cylinder size.

Closing this as explained rather than a defect to fix, setting a cylinder size/volume on your manually-logged dives should give you a correct, non-red consumption colour. Please reopen if you're still seeing an unexplained red pressure line on a dive with a cylinder size set.

closed 2026-10-17 by mikeller