subsurface/subsurface #2385: Subsurface website issues in localized pages

PeterZaal, 2019-11-15:
### Describe the issue:

- [X] Website issues
- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
There are several issues on the Subsurface website with the localized pages. These are issues I have found so far. For the localized version I mostly looked at the Dutch (nl) pages, but I have seen other languages have the same problems.

- The menu on the left side does not contain all the items in the localized version. E.g. https://subsurface-divelog.org/ contains 8 items, while https://subsurface-divelog.org/nl/ only contains 6 items. Downloads and Misc are missing, although there are localized pages for these items (/nl/download/ and /nl/misc/.
In other languages less or more links are missing, it's not all the same.
- On the dutch Download page (https://subsurface-divelog.org/nl/download/) the width of the text area is smaller than the English Download page.
- From the (English) Documentation page, the link to the FAQ is wrong. The link is to /documentation/faq, but should be /faq. Same for the localized pages, the link is to [lang]/documentation/faq, but should be [lang]/faq
- From the Documentation page, the link to the Video Tutorial is wrong. I am not sure where this should be directed to. For the localized pages the same; the link is localized, e.g. /nl/documentation/tutorial-video/, but I am not sure if there are localized videos or that it should be directed to the English video tutorial.
- The User Manual page (from Documentation) shows some literal JavaScript on the page. On the localized page the link to the User Manual is incorrect. E.g. for dutch the link is to /nl/documentation-nl/subsurface-4-gebruikershandleiding/ but should be /nl/documentation/subsurface-4-user-manual/
- The Subsurface-mobile User Manual (from Documentation) shows some literal JavaScript on the page. On the localized Download page the link to the mobile user manual links to the English mobile user manual. I am not sure if there are localized versions of the mobile user manual, but trying a localized link /nl/documentation/subsurface-mobile-v2-user-manual/ is a 404
- From the localized Misc page, e.g. /nl/misc the link to the word Privacy Policy is not translated (should be Privacybeleid), and also links to the English privacy policy page. Again not sure if there is a localized privacy policy page, because trying /nl/misc/privacy-policy/ is a 404
- (not a localized issue) Not all pages have the same width. Home, Downloads, FAQ and News have the same with, but the other items have pages with different widths. This makes the pages 'jump' when clicking around between pages.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2019-12-08:
Having so many bugs in one issue makes it a bit harder to address them one by one. I'll try to keep track here as I make progress.
I edited the issue text to make this easier.
I'll admit that finding time for this has been a major challenge with everything else that is waiting for me to work on. I'm trying to come up with a good way to change the whole mess that is our website today with something that is MUCH easier to work with. But that is turning out to be one of those projects that never ever get near the top of my queue.

dirkhh, 2019-12-08:
I fixed a couple of easy ones - but the fixes aren't permanent if I go back to my old workflow to update the pages. And that's really the scary part here. The way the pages are produced is broken.

dirkhh, 2019-12-08:
OK, I managed to make those fixes permanent. I think. Let's see what else I broke doing that.

LaPingvino, 2025-04-09:
I think as part of this issue because I'm a polyglot that has a weird collection of languages, I get a None automatically inserted that breaks the whole site...

mikeller, 2025-04-09:
@LaPingvino:

> I think as part of this issue because I'm a polyglot that has a weird collection of languages, I get a None automatically inserted that breaks the whole site...

This issue is well and truly outdated, as we have moved the Subsurface website to an entirely new framework / format in the meantime.
If you have an issue with the current format, please open a new issue and provide some more details on what the issue is. Thank you.

closed 2025-04-09 by mikeller