subsurface/subsurface #2386: Pasting cylinders only adds the first cylinder

JaspalX, 2019-11-15:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

Windows 10

Desktop

### Subsurface version:

Subsurface 4.9.3

I'm sure this has been happening on previous versions 

Release

### Steps to reproduce:

I used 12s and 7s here, but it seems to work whatever cylinder size you use

0) Need at least one dive in log, call this dive 100 set cylinders to a single 12L
1) Manually add a dive (also happens if import), call this dive 101
2) Go to equipment tab (default cylinder - single 7L - should be visible) 
3) Add another cylinder (another single 7L) (tab now shows 2x 7L)
4) apply changes
5) Ctrl-C to copy the dive (select 'cylinders')
6) Paste onto the dive from step 0 (dive 100)
7) equipment tab of dive 100 shows a single 7L cylinder only (should be 2x 7L)
optional:
8) Ctrl-C and copy cylinders from dive 100
9) Paste onto dive 101
10) Dive 101 equipment tab only shows a single 7L 
### Current behavior:

Only the first cylinder is being copied from a dive with multiple cylinders

### Expected behavior:

All the cylinders should be copied and when pasted onto another dive, all the copied cylinders should appear

### Additional information:

### Mentions:

bstoeger, 2019-11-18:
In current master, I just copied a 4-tank dive on a 3-tank dive and that seemed to work.

That raises the question: What happens if we copy a 1-tank dive onto a 3-tank dive, but all tanks were in use? I didn't dare to test that.

@atdotde: What do you think?

JaspalX, 2019-11-18:
@bstoeger if I understand correctly, you’re comment means,’it should work’ - in which case I’d be grateful for help diagnosing why it doesn’t work for me please!

bstoeger, 2019-11-19:
1) Did you try with current master? There have been rather massive changes concerning cylinder handling recently (removal of the maximum cylinder limit - support for dives without cylinders).
2) If it fails with master, could you please reproduce a minimal example that demonstrates the problem?

Thanks.

JaspalX, 2019-11-19:
Forgive my ignorance, but am I right in thinking I'd have to build the master from source in github and the version that's available on the main website (4.9.3) is not the master (that - 4.9.3 -  is the version I'm reporting the issue on).

bstoeger, 2019-11-19:
I think that there are "continuous integration" builds for master. I just don't know where. @dirkhh?

For individual pull requests you can download "artifacts", which contain builds. For example here:
https://github.com/Subsurface-divelog/subsurface/pull/2389/checks?check_run_id=308440860
If you click on artifacts you get a zip file containing a `subsurface-installer.exe` for Windows.

JaspalX, 2019-11-19:
Great, thanks. I used the installer from that link you sent and it seems to work just find copying and pasting cylinders between dives :)

bstoeger, 2019-11-19:
Cool - this was an unintentional bug fix. Perhaps we should add it to CHANGELOG.md?

dirkhh, 2019-11-19:
> I think that there are "continuous integration" builds for master. I just don't know where. @dirkhh?

The latest master CICD builds can be found at https://github.com/Subsurface-divelog/subsurface/releases/tag/ci-release
There are some known issues with the Mac binaries that make them really hard to use (and unusable on Macs not running the latest macOS) - I'm trying to come up with an alternative for that.
But the Windows installer and Linux AppImage should be fine for testing.

bstoeger, 2019-12-13:
Fix confirmed.

closed 2019-12-13 by bstoeger