subsurface/subsurface #2399: Building from sources on Debian Stretch -- fail to install dependencies

FriedrichGretz, 2019-11-23:
### Describe the issue:
- [x] Discussion request

### Issue long description:
I'm trying to build subsurface from sources. For this I follow the steps in INSTALL. It says:

> On Debian Stretch this seems to work
>
> sudo apt install \
>	autoconf automake cmake g++ git libcrypto++-dev libcurl4-openssl-dev \
>	libgit2-dev libqt5declarative5 libqt5qml5 libqt5quick5 libqt5svg5-dev \
>	libqt5webkit5-dev libsqlite3-dev libssh2-1-dev libssl-dev libssl-dev \
>	libtool libusb-1.0-0-dev libxml2-dev libxslt1-dev libzip-dev make \
>	pkg-config qml-module-qtlocation qml-module-qtpositioning \
>	qml-module-qtquick2 qt5-default qt5-qmake qtchooser qtconnectivity5-dev \
>	qtlocation5-dev qtpositioning5-dev qtscript5-dev qttools5-dev \
>	qttools5-dev-tools

There are two issues with the apt-install command:

1.) libqt5declarative5 seems to not exist in Debian stretch. At least not in mine and it is not listed under https://packages.debian.org/stretch/libs/

2.) libgit2-dev requires the installation of libcurl4-gnutls-dev. Installing the latter however conflicts with libcurl4-openssl-dev which is among the packages to be installed as in the apt-get command above.

### Operating system:
Running Linux Debian-Desktop 4.4.196 #1 SMP Tue Oct 15 16:54:21 EDT 2019 aarch64 GNU/Linux
on a Pinebook Pro

The question is: are really all of these packages needed to build subsurface and if so, how to get them?

glance-, 2019-11-23:
They were needed when that documentation where written.

I don't think qtdeclarative is needed any longer, and I don't tink it matters which ssl library your curl is linked against nowadays. GnuTLS use to be unusable back when debian started to push it for some odd interpretation of GPL, but nowadays its usable. Qt will use openssl no matter what but that doesn't really matter.

I'd suggest you look at the section a bit below named "On Raspberry Pi (Raspian Stretch and Ubuntu Mate 16.04.2) this seems to work" , and submit a PR with updated documentation when you've figured it out.

FriedrichGretz, 2019-12-09:
Thank you for the suggestion. I tried to install the packages from the other sections and that worked fine. However compilation failed because the Qt5 libraries on stretch are older than what is supported.
Thus to make it work on stretch one would also have to build Qt from source. Which is the point I decided to try out a different Linux distro on this particular machine.
So I'd consider this issue as a won't fix and therefore I do not send a RP.

closed 2019-12-09 by FriedrichGretz

dirkhh, 2019-12-09:
On Antique OSs you typically are much better off with the AppImage