subsurface/subsurface #2402: Subsurface-mobile: allow adding picutre taken with the phone and grab GPS info

Subsurface-divelog, 2019-11-25:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

This was submitted by user Karsten Hochkirch:

Ability to take a picture with your phone (most likely pre- or post- dive) and add that to the dive - then extracting the GPS information from the picture.

### Operating system:

iOS & Android

### Mentions:

@atdotde we have a good chunk of this code already, right?

Subsurface-divelog, 2019-11-25:
@olivercodes this might be fun to work on :-)

mikeller, 2026-10-18:
Hi @Subsurface-divelog,

Before committing to the new camera-integration work this needs (in-app photo capture via QML, with its own Android/iOS permissions handling), we'd like to confirm how much of the existing GPS-from-EXIF extraction logic is genuinely reusable as-is for attaching an in-app photo's location to a dive. Is this still wanted, and do you know if anyone's looked further into the reuse question since the early comment speculating "we have a good chunk of this code already"?

dirkhh, 2026-10-19:
this was an idea floated 7 years ago.
I still think this is not a bad way to deal with the "it's a pain to get mobile OSs to give us precise GPS data".
If we allow users to "import photo with precise GPS location of dive site" (and really, this could be in either desktop or mobile version), then that would be a relatively easy way for a motivated user to get to correct GPS data without having to manually copy coordinates.
This seems like a fairly easily 'vibe-codeable' task...

mikeller, 2026-10-19:
Confirmed by code inspection: core/exif.cpp and core/metadata.cpp (the EXIF/GPS parser) are already compiled into the shared core library on every platform, including mobile. So the parsing logic itself is indeed reusable as-is.

However, the piece that actually wires that parser into "attach this photo (and its GPS) to a dive" is core/picture.cpp's create_picture(), and that function is currently compiled out on mobile (#if !defined(SUBSURFACE_MOBILE)). Its only caller is the desktop-only "Load images" feature in desktop-widgets/divelistview.cpp. There is no picture-handling code anywhere in mobile-widgets/ today, so Subsurface-mobile currently has no way to attach a photo to a dive at all, on top of not extracting its GPS.

Worth noting too: even on desktop, a photo's GPS is only stored against that picture (shown as a marker on the dive map); it's not currently used to set the dive site's own coordinates. So @dirkhh's "import a photo to get a precise dive-site GPS fix" idea would be new logic on both platforms, built on top of the existing EXIF parser, that writes to the dive site location instead of (or in addition to) the picture location.

So to summarise: the hard part (EXIF GPS extraction) is solved and shared; what's missing is (a) a mobile picture-import UI, and (b) on either platform, logic to apply a photo's GPS to the dive site rather than just the picture marker.