subsurface/subsurface #2403: Import from Computer MacOS X 10.11.6 El Capitan

Ritchi78280, 2019-11-26:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
 MacOS X 10.11.6 El Capitan
 iMac 24" Intel Core2Duo

### Subsurface version:
 4.9.3
 I don't remember my previous working version of Subsurface
 Official release

### Steps to reproduce:

1) Connect a Scubapro G2 through USB to a Mac
2) Select the "Log" menu on the dive computer 
3) In Subsurface, from the Main Menu, select Import → Import From Dive Computer. The "Download from dive computer" dialogue appeared with "Vendor = Scubapro" , "Dive Computer = G2" and "Device or mount point = Scubapro - G2".
4) Select the "Download" button : Subsurface crashed

### Current behavior:

Subsurface crashed

### Expected behavior:

Subsurface imports my dive log

### Additional information:

### Mentions:

dirkhh, 2019-11-26:
I think the reason things crash is the "Core2Duo" part. I didn't realize one could upgrade a ten year old iMac to macOS 10.11. But I am 90% certain that Subsurface for a while now has included libraries that use AVX instructions which the Core2Duo doesn't support (and which will crash).
The next version of Subsurface for the Mac will require macOS 10.13 as minimum supported version.
I know this is frustrating for a user, but the amount of additional work it takes to create apps that are backwards compatible to ten year old hardware is significant...

Ritchi78280, 2019-11-26:
Hello Dirk,
Thanks for your response. 
My 2007 iMac is still alive and I very happy with it. It's been a long time since I dream of changing my iMac by a more modern, but when I see the delirium of the prices practiced by Apple, I stop thinking about it.
For your information:
I find on my Mac the "Subsurface-4.8.4.dmg" file and I launch the installation. I test an "Import from Dive computer" and it works very well :-)
I understand you don't want to support very old OS version. I'm sure you could easily do a 4.9.X release with the working librairies used in 4.8.4.  And if not, the main thing is that today, I have a working Subsurface version.
I will finish this mail by saying a big THANKS to you and all others people who develop this application

Best Regards from France
Richard

dirkhh, 2019-11-26:
Hi Richard,
I looked around and while I no longer have a Core2Duo based Mac, I do have a Mac from 2011 with ElCapitan (macOS 10.11) on it. Let me see if I can create a current version of Subsurface using that system, setting up the compile chain to not use the instruction set that isn't available on a Core2Duo.
The biggest hurdle here is that this may require a version of Qt compiled from source as I believe the official binaries already include AVX instructions.
Given all the other tasks on my todo list, I'm not sure how quickly I'll be able to make progress here...

Ritchi78280, 2019-11-28:
Hi, 

If I can help you, I will do it, like for example use ou Mac to test a beta sw build. 

In all cases, thanks for your response.

Have a good day, 

Richard

envoyé depuis mon téléphone.

dirkhh, 2020-12-12:
We support macOS 10.12 at this point, but nothing older. And I am no longer able to create binaries that run on non AVX CPUs. So I don't think I can fix this.

closed 2020-12-12 by dirkhh