subsurface/subsurface #2415: Rendering bug on web manual

glance-, 2019-12-02:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
https://subsurface-divelog.org/documentation/subsurface-4-user-manual/ gets truncated by comment about "

glance-, 2019-12-02:
This was initially reported by StrykerSKS on IRC.

PeterZaal, 2019-12-02:
I think I mentioned the same problem in
https://github.com/Subsurface-divelog/subsurface/issues/2385
(literal script in user manual page), as well as several other issues on
the website.

Peter

dirkhh, 2019-12-08:
I am WAY behind on fixing these.
And they are all signs of issues with the way I abuse WordPress to create our website.
Eventually this will all come crashing down and I'll need to completely redo this.
Today isn't that day.
I believe this bug is fixed.

closed 2019-12-08 by dirkhh