subsurface/subsurface #2418: Bug: dive list load to UI

willemferguson, 2019-12-04:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When a divelog has been loaded (xml or git; including at startup of Subsurface) the dives are shown correctly on the dive list. The last dive is (correctly) highlighted. However, the information for that dive is not populated into the tabs of the Notes panel. For instance the Dive Notes are not loaded. If one clicks on the dive in the dive list then the data for the dive are populated in the Notes panel. So it appears that, in divelist.c, the dive log is loaded correctly, the correct dive is highlighted, but that dive is actually not selected. Consequently current_dive is not initialised and its data cannot be shown.

### Operating system:

Ubuntu 8.04 running the latest Subsurface master 4.9.3-400-gf0c61

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@dirkh @bstoeger

bstoeger, 2019-12-04:
Fixed by #2400. I believe that there is no further action needed.

bstoeger, 2019-12-04:
Should be fixed in current master, please check.

willemferguson, 2019-12-04:
Fixed. Please close this issue.

bstoeger, 2019-12-04:
Thanks for confirmation.

closed 2019-12-04 by bstoeger