subsurface/subsurface #2486: Mobile:

osirisinferi, 2019-12-27:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [X] New feature request
- [ ] Discussion request

### Issue long description:

My Suunto D6i doesn't have bluetooth nor do I own an USB-A male <-> USB-C male connector, so currently I only have two ways to get my dives into Subsurface-mobile:

* by adding every dive manually (which I'm not going to do obvious, as this would take days);
* using the "Subsurface cloud".

However, I'm reluctant in using this "Subsurface cloud". Mainly, because everything in the "cloud" is a potential privacy violation. Now, "Subsurface cloud" doesn't have any privacy statement what so ever as far as I've found.

Therefore, I would like to have a *third* option: some sort of **file import** in Subsurface-mobile. I'm sure Subsurface-desktop has a fine export file format which could be transfered to your mobile device, which can be imported into Subsurface-android.

### Operating system:

Android 10 on a OnePlus 6T.

### Subsurface version:

v4.9.3 from Google Play.

### Steps to reproduce:

N/A

### Current behavior:

Impossible to import dives from a file.

### Expected behavior:

Menu option which enables user to open a file to import dives into Subsurface-mobile.

### Additional information:

N/A

### Mentions:

dirkhh, 2019-12-27:
https://lmgtfy.com/?q=Subsurface+privacy+statement&iie=1
-- 
From my phone

atdotde, 2019-12-27:
> On 27. Dec 2019, at 17:27, Dirk Hohndel <notifications@github.com <mailto:notifications@github.com>> wrote:
> 
> https://lmgtfy.com/?q=Subsurface+privacy+statement&iie=1 <https://lmgtfy.com/?q=Subsurface+privacy+statement&iie=1>
> -- 
> >From my phone

actually, for me that does not produce anything useful, but this is the glider you are looking for:

https://subsurface-divelog.org/misc/privacy-policy/ <https://subsurface-divelog.org/misc/privacy-policy/>

Best
Robert

dirkhh, 2019-12-27:
interesting - first hit, regardless of whether I'm logged in or in a private tab.
Must be a German thing :-)

Anyway, closing this issue. We have a privacy policy.
If you don't trust the policy, I would recommend not using our software because who knows what else we added to the software that spies on you.

closed 2019-12-27 by dirkhh

osirisinferi, 2019-12-27:
@dirkhh Dear god. The issue isn't about the privacy issue. That's just one of the arguments. IMHO it's a valid feature request.

dirkhh, 2019-12-27:
>**However**, I'm reluctant in using this "Subsurface cloud". Mainly, because everything in the "cloud" is a potential privacy violation. Now, "Subsurface cloud" doesn't have any privacy statement what so ever as far as I've found.

>**Therefore**, I would like to have a third option: some sort of file import in Subsurface-mobile. I'm sure Subsurface-desktop has a fine export file format which could be transfered to your mobile device, which can be imported into Subsurface-android.

Accessing files on phones is already extremely hard and will be made near impossible in Android 11. We, the Subsurface developers, have solved the problem. If you dislike our solutions, implement it (hey, this is open source), or use a different dive log.

End of conversation.