subsurface/subsurface #2491: dive planner desktop

janiversen, 2019-12-27:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When using imperial units
Bottom Po2 and Deco Po2 still shows Bar

[image]

However examining the plist file, it converts the value from psi.

### Operating system:

### Subsurface version:

Official 4.93 desktop on iMac Catalina 

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2020-01-30:
@atdotde ?

janiversen, 2020-02-05:
There seems to be little to no interest for making these changes on the desktop, so I will close this and make the adjustment for the mobile version.

closed 2020-02-05 by janiversen

atdotde, 2020-02-05:
Sorry, for the late reply: This is intentional. When we set this up, I asked around and got the astwerk that even those technical divers that use imperial units use bars for partial pressures and not psi.

janiversen, 2020-02-05:
Thanks for the explanation, I will cancel the change for mobile.

As a maybe interesting sidenote, the PADI and SSI course materials (deep dive course, which are for recreational divers, not technical divers) both use PSI in the North American version.

I guess that the BAR numbers are easier to remember than the PSI with a lot of decimals.