subsurface/subsurface #2495: Tag search in Filter Divelist doesn't show exact match but instead shows any tag that contains keyword.

devinkr, 2019-12-28:
### Describe the issue:

- [x] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
 When using the Filter divelist there is no way to do an exact search for a tag. I have both cave, and cavern tags. When I search for dives tagged cavern it works as expected. When I search for dives tagged cave it shows all dives with both cave and cavern. There is no way to only show dives tagged cave.

I actually preferred the old filter system where I could quickly see the list of tags with total count next to them and then selecting a tag would list all the dives with that tag.

### Operating system:
Windows 10 English 
Desktop
The same occurs on the mobile app searching for cave vs cavern.

### Subsurface version:

Official Release

### Steps to reproduce:
Tag dives with either cave or cavern tag. Try to search for dives tagged cave with Filter Divelist.

### Current behavior:
Currently a search for cave shows all dives tagged either cave or cavern.

### Expected behavior:
Expected behavior is that when you search for dives tagged cave it shows only those dives and not the dives tagged cavern.

### Additional information:

### Mentions:

dirkhh, 2019-12-28:
Yes, we should add an option to enforce exact match to this.

bstoeger, 2020-04-09:
I'll close this, because this should be fixed in master for quite a *long* time. Please reopen if that doesn't satisfy your needs.

That said, I think we should work towards a release, as there are many of these small fixes.

closed 2020-04-09 by bstoeger

bstoeger, 2020-04-09:
PS:

> I actually preferred the old filter system where I could quickly see the list of tags with total count next to them and then selecting a tag would list all the dives with that tag.

I agree - in my opinion the new filter was a bit rushed with fanciness valued over functionality. We could theoretically revive that feature by providing an autocomplete combo-box which lists the number of matching dives with each tag. However, that means programming these weird delegates - and I don't have a good relationship with them. When I find time, I will look into this.