subsurface/subsurface #2501: mobile: dive list is refreshed from cloud

tdyas, 2019-12-29:
### Describe the issue:
- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request
- [x] Production Issue

### Issue long description:
I am using Subsurface on desktop and mobile via cloud storage but mobile seems unable to see the updated cloud storage even if I pull to refresh.

I downloaded a dive into my desktop and saved it to the cloud storage. (The title bar of the desktop instance says "cloud storage.") I then started the mobile app on my iPhone. I first noticed that it did not get the new dive from the cloud storage even after I pulled to refresh the dive list.

### Operating system:
iOS 

### Subsurface version:
Desktop v4.9.3 on Ubuntu 19.10 (via snap)
Mobile v2.2.3(4.9.3.273) on iOS 13.3 on iPhone 7 hardware

### Steps to reproduce:

### Current behavior:
See long description.

### Expected behavior:
Dives added from desktop are loaded from cloud storage on mobile

dirkhh, 2019-12-30:
first thing to check is if the dives were successfully synced to the cloud. Go to 

https://cloud.subsurface-divelog.org

and log in using your cloud credentials. Are the dives there? 

If yes, then please try and refresh the dive list on your iDevice and after you have done this, tap on "Ask for support" in the main menu and send me an email (it opens the email app with logs attached) referencing this issue (just issue 2501 - you don't need to add a link or anything).

If no, then the problem is on your PC. Please run Subsurface from the command line with the verbose option ('-v') and redirect the output to a file. Try to again save to the cloud and then attach the log file to this issue.

tdyas, 2019-12-31:
The cloud storage has all of the dives. The "Ask for support" option tries to invoke the device's Mail app but I don't use that app (and would have to configure it).  I turned on the Developer menu and found the Application Log option. (I wish I had found this sooner!) Here's the log excerpt (hand typing it since the log's UI doesn't offer a copy to clipboard option): 

```
"0.675: Synchronising data file"
"0.682: Load dives from local cache"
"0.690: Successfully opened dive data"
"0.696: 31 dives loaded from cache"
"0.697: have cloud credentials, but user asked not to connect to network"
```

Then the log just goes into pixel size stuff. 

I don't recall setting it to not sync. The interesting thing is when I go to the Settings page, the "Status" under the "Cloud status" is "Undefined." And then if I click the `Change` button, the message pane in the middle just says "Starting..." I can 

I tried to sign in again and when I do the log then gets these messages (with my email redacted):

```
"1132.675: verify credentials for email XXXXXX (no PIN)"
"1133.177: PIN accepted"
```
But still no change in sync'ing the new dives (and no cloud storage messages after that PIN accepted message).

The "Cloud credentials" is then in the "Verified" status but going to the Change screen still shows that "Starting..." message. (I can email a screenshot if you'd like.) And then when I go back to settings, it is at "undefined" status again.

Tangentially, there is an interesting TypeError in the log:

```
"1137.896: pageStack switched to DiveList"
<Unknown File>:7:TypeError: Cannot read property 'contentX' of null
```

tdyas, 2019-12-31:
Found the in-app email in the source. Emailed screenshots of the logs and the Cloud Settings page.

dirkhh, 2019-12-31:
Nothing magic about the in app email address - in the end it just goes to me (because we get so much rather frustrating email to that alias that I don't want to expose others to it).

To copy the full log (and I'd really need to see the full log) simply go to About screen and tap the button on that screen that copies it to the clipboard. Then you can paste it into any app you want.

tdyas, 2019-12-31:
> To copy the full log (and I'd really need to see the full log) simply go to About screen and tap the button on that screen that copies it to the clipboard. Then you can paste it into any app you want.

I had tried that originally but it wouldn't paste into Gmail at the time I was opening this issue (and still doesn't). I just tried pasting it into the built-in Notes app and then re-copied it to clipboard and it now pastes into the Gmail app just fine. You should see it in email.

dirkhh, 2019-12-31:
That is strange. Works here on a somewhat older iPhone 8 plus...

Got your email and responded

tdyas, 2019-12-31:
Resolved on the email discussion. (I ended up using the "Enable cloud auto sync" and then restarting the app.)

closed 2019-12-31 by tdyas