subsurface/subsurface #2510: Android app BT download from Shearwater Petrel does not work

Laarzmo, 2020-01-01:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When downloading dives from Petrel via bluetooth everything seems to go ok but after the communication do dives appear.

### Operating system:

Android 10

Oneplus 7 pro / Sherwater Petrel firmware v72

### Subsurface version:

Subsurface-mobile:2.2.3(4.9.3.273)

Official release

### Steps to reproduce:

Select dive management
-> Download from DC
[image]

### Current behavior:

No dives downloaded

### Expected behavior:

### Additional information:

[image]

That "Error parsing the header" & "Error parsing the datetime" repeats for all 33 dives which can be found in DC
And it end "No new dives downloaded from dive computer"
### Mentions:

dirkhh, 2020-01-30:
sorry, this one slipped through the cracks.
Which firmware version are you running on that Petrel?  In the log above it identifies itself as "72" - and the newest I have seen is firmware 7.1 - is this beta firmware?
I have both a Petrel and a Petrel 2 and have never seen this, but I am still on firmware version 6.5. And there is a chance that they changed things in the communications protocol and forgot to let us know (they are usually very good about this).

PeterZaal, 2020-01-30:
I heard about the v72 firmware about a month ago. It was also discussed on
a GUE CCR group on Facebook. Some people had problems with very short
battery life when using it on a DiveCan handset and were told to check and
update their firmware. Back then the v72 firmware was offered via the
Shearwater Desktop software, but it was not for download on the website.
Which was a bit weird, so I didn't upgrade my Petrel 2 then. Just checked
their website again and the v72 is now also available for download, so v72
is now officialy the latest version.
Their release notes state there are no functional changes, just support for
minor production changes. So I guess the log format didn't change, just a
new version number?

Laarzmo, 2020-01-30:
> sorry, this one slipped through the cracks.
> Which firmware version are you running on that Petrel? In the log above it identifies itself as "72" - and the newest I have seen is firmware 7.1 - is this beta firmware?
> I have both a Petrel and a Petrel 2 and have never seen this error, but I am still on firmware version 6.5. And there is a chance that they changed things in the communications protocol and forgot to let us know (they are usually very good about this).

Hi,

Yes I can confirm that this is on firmware v72.
I have not tried to load my dives from DC with any other version.

Is there a possibility to grap more detailed trace from the download process?

dirkhh, 2020-01-30:
> On Jan 30, 2020, at 7:50 PM, Laarzmo <notifications@github.com> wrote:
> 
> sorry, this one slipped through the cracks.
> Which firmware version are you running on that Petrel? In the log above it identifies itself as "72" - and the newest I have seen is firmware 7.1 - is this beta firmware?
> I have both a Petrel and a Petrel 2 and have never seen this error, but I am still on firmware version 6.5. And there is a chance that they changed things in the communications protocol and forgot to let us know (they are usually very good about this).
> 
> Hi,
> 
> Yes I can confirm that this is on firmware v72.
> I have not tried to load my dives from DC with any other version.
> 
> Is there a possibility to grap more detailed trace from the download process?
> 

I just heard back from Shearwater (they really are great to work with). They tell me that there were no log format changes. So your errors are even more strange.
I assume you can reproduce them, correct?

Unfortunately, IIRC the Petrel doesn't allow us to do a full dump of its memory, so we'd need to get a packet level log to see what's going on.

Before I get too far into the weeds here, Jef, any suggestions?

/D

dirkhh, 2020-01-30:
On 30/01/2020 20:45, Dirk Hohndel wrote:
> Unfortunately, IIRC the Petrel doesn't allow us to do a full dump of its memory, 
> so we'd need to get a packet level log to see what's going on.
> 
> Before I get too far into the weeds here, Jef, any suggestions?
Sure. The problem appears to be in the parsing. So we don't really need a full 
memory dump, only the raw data of each dive.

The easiest way to get that on a desktop PC is with the libdivecomputer dctool 
command-line tool:

./dctool -vv -l petrel.log -f petrel download -o dive.%n.bin -f raw -t bluetooth

Unfortunately this only works for Bluetooth classic (not BLE) on Windows and 
Linux (not Mac).

The alternative is to simply do a normal download with subsurface (desktop or 
mobile), with the libdivecomputer logfile enabled. I have a debug tool to replay 
such logfiles, so I can reproduce and debug the problem locally. So that log 
file is what I need to look into this.

In subsurface mobile the logs are available from the about screen if I remember 
correctly.

Jef

Laarzmo, 2020-02-15:
Hi,

Sorry about long delay.
But now I finally managed to do something about this.

I installed windows version about the subsurface and it downloaded the dives.
I tried to create a dump as well but when trying that I got "Dive data input error" when I used classic mode and "Unable to open LE:xxxx" when trying with LE mode.

In Android version I don't have dump option during download.

Br,
Lauri Liehu

dirkhh, 2020-02-15:
Shearwater dive computers don't support creating a dump file.
Also, we use the exact same code to download and parse the dives on Android and on Windows, so this outcome is very confusing to me.

We have seen many problems connecting to the dual interface versions of the Petrel - but the parsing error you first reported is not consistent to that underlying issue. I would have expected the download to simply not start on the Android device.

dirkhh, 2020-12-12:
I am able to download both from a Petrel and a Petrel 2 here with the current app on Android. Closing this until I see a new logfile that shows an issue.

closed 2020-12-12 by dirkhh